google-deepmind/mujoco #1516: [MJX] Question about batched data

Gregwar, 2024-03-15:
Hello,

I am trying to understand how to batch simulations using XLA.

Here is an example of simulation using one ball from the tutorial:
https://gist.github.com/Gregwar/b1b7462d6f9ae144ffa70e28390b56dc#file-one_ball-py

As a result:
```
Time elapsed: 1.2000079154968262
Time elapsed: 0.0005218982696533203
Time elapsed: 0.00037169456481933594
Time elapsed: 0.0003247261047363281
```

Here is an example of simulation using a batch of N=1 ball, still from the tutorial:
https://gist.github.com/Gregwar/b1b7462d6f9ae144ffa70e28390b56dc#file-one_ball_1batch-py

And now the result becomes:
```
Time elapsed: 8.233286619186401
Time elapsed: 0.42046070098876953
Time elapsed: 0.40743041038513184
Time elapsed: 0.41671061515808105
```

It looks like something is really really bad when using batched MjxData here, but I feel like I mostly assembled code from the tutorial. How come a batch of 1 object can be this much slower than one object alone ?

erikfrey, 2024-03-15:
Hi @Gregwar - it looks like you've run into a tricky JAX gotcha.  Wrapping a function in `jax.vmap` does not JIT, so each time you call `jit_step` in your second example, you are running unjitted code.  Please try this:

```py
jit_step = jax.jit(jax.vmap(mjx.step, in_axes=(None, 0)))
```

Hope this helps!

closed 2024-03-15 by erikfrey

Gregwar, 2024-03-15:
Thanks!

Maybe this can be updated accordingly:
https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=Nj4-Xmx4DFaq

The name `jit_step` leads to this confusion I guess:
```python
...
jit_step = jax.vmap(mjx.step, in_axes=(None, 0))
...
```

Gregwar, 2024-03-15:
For the records, I made a MJX benchmarking script to compare with CPU counterpart `testspeed`
https://gist.github.com/Gregwar/a5e363e38c282cca9577095a70f36dc9

Do not hesitate if you see anything wrong

erikfrey, 2024-03-15:
Good point - addressed in e95159b4f6d48d114b16a8dc13ad26b3e44bc3e2.