google-deepmind/mujoco #1517: Nan jacobian of mjx.step while in contact

Andrew-Luo1, 2024-03-17:
## Issue

By the jacobian of mjx.step, I refer to the matrix J such that the ith row and jth column is $\frac{\partial s_{k+1, i}}{\partial d_{k, j}}$, where $s_{k+1}$ is the state at time k+1 and $d_k$ is the concatenation of $s_k$ and control $a_k$. If m is an mjx.model, J has m.nq + m.nv rows and m.nq + m.nv + m.nu columns.

When simulating a geom leaving the ground, jax.jacfwd returns nan gradients. I was wondering if this is expected behaviour, or a bug? While in the case of hard contact, a nan would make sense, I am confused because mujoco uses soft contacts.

The motivation for this ticket is that I often get nan gradients for rewards calculated from states with respect to the parameters of (neural network) policies that produce controls driving the states. I have found that the nan policy gradients are associated with these jacobians, since they're part of the computation graph.

## Code
```
import mujoco
import mediapy as media
from mujoco import mjx
import jax
from jax import numpy as jp

box_xml = """ 
<mujoco>
  <option timestep="0.002" iterations="10" ls_iterations="10" solver="Newton"/>
  <option cone="pyramidal" impratio="100"/>

  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3"
     rgb2=".2 .3 .4" width="300" height="300"/>
    <material name="grid" texture="grid" texrepeat="8 8" reflectance=".2"/>
  </asset>

  <worldbody>
      <light pos="1 1 2" mode="targetbodycom" target="box"/>
      <camera name="target" pos="0 -3 2" xyaxes="1 0 0 0 1 2" mode="targetbodycom" target="box"/>

      <geom size="1 1 .01" type="plane" material="grid"/>

      <body name="box" pos="0 0 0.5">
      <joint name="slider" axis="0 0 1" type="slide"/>
      <geom name="box" type="box" size="0.5 0.5 0.5" condim="3" friction="0.8 0.02 0.01" priority="1" solimp="0.015 1 0.031"/>
      </body>
  </worldbody>

  <actuator>
    <general name="box_pusher" joint="slider" gaintype="fixed" gainprm="1"/>
  </actuator>

</mujoco>
"""

def step(diffwrt, dobj, sys):

    dobj = dobj.tree_replace({
        'qpos': diffwrt[:sys.nq],
        'qvel': diffwrt[sys.nq:sys.nq+sys.nv],
        'ctrl': diffwrt[sys.nq+sys.nv:]
    })

    dobj = jmjxstep(sys, dobj)

    state = jp.squeeze(jp.concatenate(
                           [jp.expand_dims(dobj.qpos, 1), 
                            jp.expand_dims(dobj.qvel, 1)],
                            axis=0))
    return state

jacstep = jax.jacfwd(step, 0)
jjacstep = jax.jit(jacstep)
jmjxstep = jax.jit(mjx.step)

model = mujoco.MjModel.from_xml_string(box_xml)
data = mujoco.MjData(model)

num_steps = int(2*500)
ctrl_max = 1e4
dt = model.opt.timestep
ctrls = jp.linspace(ctrl_max/2, ctrl_max, num_steps)

mjx_model = mjx.device_put(model)
mjx_data = mjx.device_put(data)

nan_jacs = 0
for i in range(num_steps):
    mjx_data = mjx_data.tree_replace({
        'ctrl': ctrls[int(mjx_data.time/dt)]
    })
    mjx_data = jmjxstep(mjx_model, mjx_data)

    ctrl_2_cat = mjx_data.ctrl.reshape((jp.size(mjx_data.ctrl), 1))
    x_i = jp.squeeze(jp.concatenate(
            [jp.expand_dims(mjx_data.qpos, 1), 
            jp.expand_dims(mjx_data.qvel, 1),
            ctrl_2_cat], axis=0))

    cur_jac = jjacstep(x_i, mjx_data, mjx_model)
    if jp.sum(jp.isnan(cur_jac)) == cur_jac.size:
        nan_jacs += 1
    
print(f"{nan_jacs} nan jacobians")
```
On my setup, this prints `228 nan jacobians`.

## Video
Ground reaction forces shown with blue arrows. Box flashes red upon nan jacobian.

[image]

## Setup
Operating System: Ubuntu 20.04
MuJoCo & MJX version: 3.1.3

btaba, 2024-03-26:
Hi @Andrew-Luo1 you shouldn't be getting NaNs. Please use the following debug flags, and report back:

Debug NaN:

```
from jax import config
config.update("jax_debug_nans", True)
```

Is it a precision issue?

`jax.config.update("jax_enable_x64", True)`
`jax.config.update('jax_default_matmul_precision', jax.lax.Precision.HIGH)`

Andrew-Luo1, 2024-03-29:
Hi @btaba,

When I turn on jax_debug_nans, on the line ```cur_jac = jjacstep(x_i, mjx_data, mjx_model)``` I get the error message ```Exception has occured: FloatingPointError```, and error continues on to say that the same error doesn't happen for the deoptimized version. 

Turning on those two flags fixes the nan so it's probably a precision issue. Thanks!

closed 2024-03-29 by Andrew-Luo1