google-deepmind/mujoco #1518: Memory of `list[mjData]` is not cleared after calling `del`

milutter, 2024-03-18:
When deleting a python `list` containing `n` `mjData` objects the memory does not seem to be cleared (as one would expect). In pseudo-code:
```
some_list = [mujoco.MjData(mj_model) for n in range(N)]  # Memory increases 
del some_list  # Memory does NOT decrease
``` 
* Is this the expected behavior?
* Are there references inside `mjModel` to all derived `mjData` preventing freeing the memory? 
* If this is the expected behavior, is there a way to clear that memory?

I am no pybind expert and so don't have a deep understanding of the pybind garbage collection / object lifetime. When doing the same experiment with larger `numpy` arrays instead of `mjData` objects, the memory is cleared. A minimal example and an example comparing different variations is contained below.

Further observations: 
* When overwriting the `mj_data` variable in a `for` loop the memory consumption does not accumulate.

* The increase in memory is not linear when creating and deleting a new list of `mjData`. However, there seems to be a continuous increase.

* Interestingly, for `numpy` clearing the memory works. However, after calling `del list[mjData]` once, the memory for the `numpy` lists is also not cleared anymore. This observation could be a red herring.

- System:
  - Ubuntu 22.04 / Python 3.10
  - MuJoCo version: 3.1.3 (installed via pip)
  - Python bindings: `mujoco`

#### Example Output:
```
# Minimal Example: 
Process Memory: 47.41MiB
Process Memory: 1822.62MiB
Process Memory: 1782.35MiB

# Elaborate Example:
(000)      initialize - Process Memory =    45.36 MiB -> Delta =        +0.00 MiB
(001)         mjModel - Process Memory =    47.46 MiB -> Delta =        +2.11 MiB

(002)   Create mjData - Process Memory =    48.67 MiB -> Delta =        +1.20 MiB
(003)      Del mjData - Process Memory =    48.73 MiB -> Delta =        +0.06 MiB

(004)      numpy List - Process Memory =  1072.62 MiB -> Delta =     +1023.89 MiB
(005)        del List - Process Memory =   110.94 MiB -> Delta =      -961.68 MiB
(006)            Wait - Process Memory =   110.94 MiB -> Delta =        +0.00 MiB

(007)      numpy List - Process Memory =  1072.83 MiB -> Delta =      +961.89 MiB
(008)        del List - Process Memory =   110.94 MiB -> Delta =      -961.89 MiB
(009)            Wait - Process Memory =   110.94 MiB -> Delta =        +0.00 MiB

(010)     mjData List - Process Memory =  1872.31 MiB -> Delta =     +1761.37 MiB
(011)        del List - Process Memory =  1832.04 MiB -> Delta =       -40.27 MiB
(012)            Wait - Process Memory =  1832.04 MiB -> Delta =        +0.00 MiB

(013)     mjData List - Process Memory =  2635.63 MiB -> Delta =      +803.60 MiB
(014)        del List - Process Memory =  2595.33 MiB -> Delta =       -40.30 MiB
(015)            Wait - Process Memory =  2595.33 MiB -> Delta =        +0.00 MiB

(016)     mjData List - Process Memory =  2717.27 MiB -> Delta =      +121.94 MiB
(017)        del List - Process Memory =  2677.15 MiB -> Delta =       -40.12 MiB
(018)            Wait - Process Memory =  2677.15 MiB -> Delta =        +0.00 MiB

(019)      numpy List - Process Memory =  3628.47 MiB -> Delta =      +951.32 MiB
(020)        del List - Process Memory =  3628.47 MiB -> Delta =        +0.00 MiB
(021)            Wait - Process Memory =  3628.47 MiB -> Delta =        +0.00 MiB
```

#### Example Code:
```
# Standard Library
import gc
import pathlib
import time

# Third Party
import mujoco
import numpy as np
import psutil

MUJOCO_XML = """
<mujoco model="cart-pole">
  <option timestep="0.01" integrator="RK4">
    <flag contact="disable" energy="disable"/>
  </option>

  <asset>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2" width="100" height="100"></texture>
    <material name="MatPlane" reflectance="0.5" texture="texplane" texrepeat="1 1" texuniform="true"></material>
  </asset>

  <default>
    <default class="pole">
      <joint type="hinge" axis="0 1 0"  damping="2e-6"/>
      <geom type="capsule" fromto="0 0 0 0 0 1" size="0.045" mass=".1"/>
    </default>
  </default>

  <worldbody>
    <light name="light" pos="0 0 6"/>
    <geom name="floor" pos="0 0 -.05" size="4 4 .2" type="plane" material="MatPlane"/>
    <geom name="rail1" type="capsule" pos="0  .07 1" zaxis="1 0 0" size="0.02 2"/>
    <geom name="rail2" type="capsule" pos="0 -.07 1" zaxis="1 0 0" size="0.02 2"/>
    <body name="cart" pos="0 0 1">
      <joint name="slider" type="slide" limited="true" axis="1 0 0" range="-1.8 1.8" solreflimit=".08 1" damping="5e-4"/>
      <geom name="cart" type="box" size="0.2 0.15 0.1" mass="1"/>
      <body name="pole_1" childclass="pole">
        <joint name="hinge_1"/>
        <geom name="pole_1"/>
      </body>
    </body>
  </worldbody>

  <actuator>
    <motor name="slide" joint="slider" gear="10" ctrllimited="true" ctrlrange="-1 1" />
  </actuator>
</mujoco>
"""

######################################################
######### Minimal example

def minimal_example():
    print('\nMinimal Example:')
    mj_model = mujoco.MjModel.from_xml_string(MUJOCO_XML)
    print(f"Process Memory: {psutil.Process().memory_info().rss / 1024**2:.2f}MiB")

    a_list = [mujoco.MjData(mj_model) for _ in range(4096)]
    print(f"Process Memory: {psutil.Process().memory_info().rss / 1024 ** 2:.2f}MiB")

    del a_list
    gc.collect()
    print(f"Process Memory: {psutil.Process().memory_info().rss / 1024 ** 2:.2f}MiB")

######################################################
######### In-depth example with numpy comparison

class MemoryTracker:
    """Just a pretty print function for process memory."""

    def __init__(self):
        self.counter = 0
        self.memory = self.get_process_mem()
        self._print(self.counter, 'initialize', self.memory, 0)

    @staticmethod
    def get_process_mem(pid=None):
        return psutil.Process(pid).memory_info().rss

    @staticmethod
    def _print(counter, key, memory, delta):
        print(f"({counter:03d}) {key:>15} - "
              f"Process Memory = {memory / 1024 ** 2:8.2f} MiB "
              f"-> Delta = {delta/1024 ** 2:+12.2f} MiB")

    def step(self, key):
        self.counter += 1
        new_memory = self.get_process_mem()
        self._print(self.counter, key, new_memory, new_memory - self.memory)
        self.memory = new_memory

def elaborate_example():
    print('\nMore elaborate Example:')
    n = 4096
    tracker = MemoryTracker()

    def list_test(key, list_constructor):
        """This test function:

        1) constructs lists via passed function
        2) records process memory
        3) deletes lists & calls garbage collection
        4) records process memory
        5) sleeps
        6) records process memory
        """
        print('')
        a_list = list_constructor()
        tracker.step(f'{key} List')

        del a_list
        gc.collect()
        tracker.step('del List')

        time.sleep(1.0)
        gc.collect()
        tracker.step('Wait')

    mj_model = mujoco.MjModel.from_xml_string(MUJOCO_XML)
    tracker.step('mjModel')

    #####
    # Overwrite MjData in a loop:
    print()
    mj_data = None
    for i in range(n):
        mj_data = mujoco.MjData(mj_model)

    tracker.step('Create mjData')
    del mj_data
    gc.collect()
    tracker.step('Del mjData')

    #####
    # Delete list of large numpy arrays:
    numpy_list_fn = lambda: [np.random.uniform(size=(256, 128)) for _ in range(n)]
    list_test('numpy', numpy_list_fn)
    list_test('numpy', numpy_list_fn)

    #####
    # Delete list of mjData
    mj_data_list_fn = lambda: [mujoco.MjData(mj_model) for _ in range(n)]
    list_test('mjData', mj_data_list_fn)
    list_test('mjData', mj_data_list_fn)
    list_test('mjData', mj_data_list_fn)

    #####
    # Delete list of large numpy arrays:
    list_test('numpy', numpy_list_fn)
```

nimrod-gileadi, 2024-03-18:
Thanks for the detailed report.

I have been able to reproduce the issue in a test, but I also verified (by changing the bindings to print out some log lines) that for each created mjData, mj_deleteData is called as expected, so these aren't actually mjData objects that are getting leaked.

If I run:

```py
for _ in range(1000):
    a_list = [mujoco.MjData(mj_model) for _ in range(4096)]
    print(f"Process Memory: {psutil.Process().memory_info().rss / 1024 ** 2:.2f}MiB")

    del a_list
```

I notice that memory does increase steadily, until at some point it flatlines and stops growing, which you wouldn't expect if this was a normal leak. I think this is to do with how the C Python implementation decides to release memory back to the OS. If you search around Stack Overflow you'll find a few similar issues. I liked this short post: https://web.archive.org/web/20110621221104/http://effbot.org/pyfaq/why-doesnt-python-release-the-memory-when-i-delete-a-large-object.htm

milutter, 2024-03-19:
Hi Nimrod, thanks for looking into this. I also noticed that the increase in memory is sub-linear, which would not be possible if memory would be leaked. It is a bit annoying that C Python is so unpredictable in this regard. In more memory constrained use-cases, this can be a bit more challenging as one can run out of memory before the memory levels off. Most likely though one can design around this problem for most use-cases.

closed 2024-07-11 by nimrod-gileadi