google-deepmind/mujoco #1520: Not able to use all MuJoCo scripts from the plug in for Unity

yaama19, 2024-03-18:
Hi,

I'm an engineer currently working wit MuJoCo and I'm trying to use MuJoCo for simulate the physics. I'm working wit the OS Windows. 

I'm looking for some help with the  implementations and how to use the scripts given by the unity plug in zip . I have add the package and I am able to use some Scripts as "MJ Geom" or "Mj Mesh Filter" but then I'm not able to use the "Mouse spring" and I can't add a component with he related script. And in the unity plug in zip there's a lot of scripts aviable to use and I can't use them. As Iong as I know the implementation is correct, I add the human body MuJoCo scene, I don't know what could be possible wrong. I also allowed to use the unsafe scripts in the global configuration.

Any help is accepted! 

Thnaks a lot.

Balint-H, 2024-03-18:
Hello,

Thanks for your interest in the Unity plugin! Here's a tutorial on using the plugin. It was made with a focus on RL applications but the first couple tutorials are relevant for general use cases as well:

https://github.com/Balint-H/mj-unity-tutorial

Can you tell me what is your target use for the plugin? I'll be able to give more specific advice with that.

Balint-H, 2024-03-19:
Hi,

I forgot to re-enable the welcome dialog box in the tutorial. I apologize, it should be fixed now and it should display the following info when you first open the "MuJoCo-Unity-Tutorial" project with Unity Hub:

[image]

Clicking `Tutorials>Welcome Dialog` on the toolbar will also display this message.

The tutorials should be permanently visible docked on the right side of the editor window like this (If they are not there click `Tutorials>Show Tutorials` on the toolbar`:

[image]

Clicking on one of them should launch the tutorial, and let you navigate it while in the editor:

[image]

Please let me know if you need any further guidance!

yaama19, 2024-03-19:
Thanks you so much, I have just done it. But I still have the problem, I'm not able to move de scene (pole), I' m just able to move the pole when I'm in the play mode, not like the tutorial says, I think maybe something is wrong with the files? I attached you in the picture that I'm only moving the slide joint not the gameObject itself, then I can't see the physics movement of the cart/pole.
I don't know if it's related but at the begging when I first opened the project, the console showed a problem: "**_Assets\Tutorials\3 - RL Basics\RL Scripts\SlideObservations.cs(64,32): error CS0246: The type or namespace name 'MjSlideJoint' could not be found (are you missing a using directive or an assembly reference?)_**" 
 And then I checked the package Manager and the MuJoCo package wasn't update, so I add the one it's in GitHub made by deepmind. And the error was solved, but maybe has something to do with the movement problem.

I wait for your answer! and thank you for your time, 
 
[image]

Balint-H, 2024-03-19:
The project references the 3.1.3 version of the MuJoCo plugin, and it should automatically fetch when you first open it. Even after a fresh install of the tutorial and loading it for the first time, you should be able to see the MuJoCo package in the "Packages" folder in the Project tab (bottom left). 

When opening the project from Unity Hub, are you selecting the folder `mj-unity-tutorial` or  the subfolder `Mujoco-Unity-Tutorial` (correct one)?

Once you enter play mode, the MuJoCo runtime is responsible for determining the motion of the model. The only way to control the system is through interacting with the MuJoCo runtime, directly setting the Unity transforms won't work. While outside play mode, you can move and adjust the gameobjects as you normally would in Unity.

Does the cartpole move when you adjust the "Control" parameter of the actuator (after you entered play mode)?
You can also apply forces to the system via the Mouse Spring interface. Click on a shape in the Scene view while in play mode, then perform Ctrl+Left click drag to apply forces (or Ctrl+Shift+Left click drag for movement along an orthogonal plane).

To confirm your project is set up correctly you can run the scene 
`mj-unity-tutorial\MuJoCo-Unity-Tutorial\Assets\Tutorials\3 - RL Basics\Scenes\CartPoleTutorialTrained.unity`
It should give you the behaviour as below (I ran it from a fresh install just in case):

https://github.com/google-deepmind/mujoco/assets/41113387/36d87048-6577-458c-b5a7-15cf0e22ddc4

yaama19, 2024-03-20:
Hi again, I think now it's working :)
Yes! I'm opening the correct file. 
When I first open the project without doing anything else I have this problem:
[image]

Looks like it doesn't detect a package and when I try to do it manual, it says to use a a correct file (as you can see in the console window in the previous picture). And in the package Manager just show this:
[image]

Also when I first open the project, the MuJoCo folder in the Packages doesn't show. 
[image]

Of course I can't run the project with the problems, so I add the MuJoCo package.json, the one I mention in the previous comments, the one I extracted from DeepMind in GitHub. Then I can run the program. I don't know what was wrong, but at least  it's working. 
Eventho I would like to know more about this implementation. I have seen in a video that there are more scripts available, but I can find them. For exemple a "Add a prefab to scence" (like in the picture). It's an interesting function, and just like this one, I also wold like to implement the " Mosue spring" script, but again I can find it.

[image]

Anyway with the things you told me till now it's more than enought! 
I am so grateful with you, thanks a lot.

Balint-H, 2024-03-20:
Ah, I see the issue! The MuJoCo package is referenced through the git URL of the package. When Unity tries to download it, it will search for git on your system, to download it with. However, either git is not installed on your system or Unity can't see it. For this reason the package fetching fails. 

Do you have git installed (https://git-scm.com/download/win)? You can download the package as a zip from GitHub as well, and add the package locally, alternatively, but then you have to do that manually every time you install the project or want to update it. I'll mention that git is required to automatically fetch the package on the tutorial's Readme in the future.

MouseSpring is implemented in the Editor, it doesn't have to be added to your scene. With the package installed, you can just click on a mujoco body in the Scene view (not in the Game view! If you need it there let me know, we'd need to implement a component for that).

The AddPrefabToScene component is not specific to the MuJoCo plugin. It appears to me to be a custom script written by Tom to demonstrate the prefab instantiation works with MuJoCo prefabs. For a guide and example scripts on how to write your own component that does that see the Unity manual: https://docs.unity3d.com/Manual/InstantiatingPrefabs.html

yaama19, 2024-03-22:
Hi! 
Yes! now it's working. The problem may be the git thing, I didn't have it install. About the Mouse spring thank you too, I was doing it in a bad way, I was looking at the scripts in the unity-mujoco plug in and I expected to be able to add all the scripts I saw there in any unity gameObjeect. 

I will check the link and try to do the script by myself. 

Thank you so much again!

closed 2024-03-27 by erez-tom