google-deepmind/mujoco #1521: Incorporating additional objects into an existing environment

HarshithaBR, 2024-03-18:
**Incorporating particles.xml into an existing environment**
Hi,

We are students trying to use MuJoCo to simulate an environment with additional elements.

We intend to integrate a modified version of the composite <particle.xml> environment.

1. Incorporate the contents of the xml file shown below:
This is a modification of the "particle.xml" composite objects.

 Modified [particles.xml](https://github.com/google-deepmind/mujoco/blob/main/model/composite/particle.xml) -
<mujoco model="Particle">

  <include file="scene.xml"/>

  <option solver="CG" tolerance="1e-6" timestep=".01"/>

  <size memory="1G"/>

  <visual>
    <map stiffness="100"/>
  </visual>

  <default>
    <default class="wall">
      <geom type="plane" size=".5 .5 .05"/>
    </default>
  </default>

  <worldbody>
    <geom name="box_1" type="box" size=".3 .3 .3" pos=".6 0 0.15"/>
    <geom name="box_2" type="box" size=".3 .3 .3" pos="-.6 0 0.15"/>
    <geom name="box_3" type="box" size=".3 .3 .3" pos="0 .6 0.15"/>
    <geom name="box_4" type="box" size=".3 .3 .3" pos="0 -.6 0.15"/>
    <composite type="particle" count="2 1 1" spacing="0.07" offset="0 0 1">
      <geom size=".025" rgba=".8 .2 .1 1"/>
    </composite>
  </worldbody>
</mujoco>

<mujoco model="ParticleOverflow">
    <include file="scene.xml"/>
    
    <option solver="CG" tolerance="1e-6" timestep=".01"/>
    
    <size memory="1G"/>
    
    <visual>
        <map stiffness="100"/>
    </visual>
    
    <default>
        <default class="wall">
            <geom type="plane" size=".5 .5 .05"/>
        </default>
    </default>
    
    <worldbody>
        
        <geom name="central_box" type="box" size=".5 .5 .5" pos="0 0 0.25"/>
        
        
        <geom name="box_1" type="box" size=".03 .03 .03" pos=".6 0 0.15"/>
        <geom name="box_2" type="box" size=".03 .03 .03" pos="-.6 0 0.15"/>
        <geom name="box_3" type="box" size=".3 .3 .3" pos="0 .6 0.15"/>
        <geom name="box_4" type="box" size=".3 .3 .3" pos="0 -.6 0.15"/>
        
        
        <composite type="particle" count="10 10 20" spacing="0.03" offset="0 0 1">
            <geom size=".025" rgba=".8 .2 .1 1"/>
        </composite>
    </worldbody>
</mujoco>

into an [existing environment](https://github.com/MarkFzp/act-plus-plus/blob/main/assets/bimanual_viperx_ee_transfer_cube.xml) :

<details>
<summary> Addition to the above existing environment would be the following world body (as taken from the code linked above)</summary>

<worldbody>
        <include file="vx300s_left.xml" />
        <include file="vx300s_right.xml" />

        <body name="box" pos="0.2 0.5 0.05">
            <joint name="red_box_joint" type="free" frictionloss="0.01" />
            <inertial pos="0 0 0" mass="0.05" diaginertia="0.002 0.002 0.002" />
            <geom condim="4" solimp="2 1 0.01" solref="0.01 1" friction="1 0.005 0.0001" pos="0 0 0" size="0.02 0.02 0.02" type="box" name="red_box" rgba="1 0 0 1" />
        </body>
        <geom name="box_1" type="box" size=".03 .3 .3" pos=".6 0 0.15"/>
        <geom name="box_2" type="box" size=".03 .3 .3" pos="-.6 0 0.15"/>
        <geom name="box_3" type="box" size=".3 .3 .3" pos="0 .6 0.15"/>
        <geom name="box_4" type="box" size=".3 .3 .3" pos="0 -.6 0.15"/>
        <composite type="particle" count="2 1 1" spacing="0.03" offset="0 0 1">
            <geom size=".025" mass="0.6" rgba=".8 .2 .1 1"/>
        </composite>
    </worldbody>

</details>

We are able to get the objects in the environment by including them into the xml as shown above, however we are facing errors corresponding to properties of mass and inertia, when we try to simulate the environment by passing path of this xml to the command: `./simulate <path to the xml file>`

The error we came across when we  is as follows:

`Error: mass and inertia of moving bodies must be larger than mjMINVAL
Object name = B0_0_0, id = 26`

What is the best way to go about with this?

Thanks and Regards

yuvaltassa, 2024-03-19:
You are including a file with the string `<compiler ... inertiagrouprange="4 5"/>`. This is telling the compiler to [only infer inertias from geoms](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-inertiagrouprange) in groups 4 and 5. [The default group for geoms (and everything else) is 0](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-group). So the `mass="0.6"` inside your composite is being ignored and you have a massless moving body, which is not allowed.

closed 2024-03-19 by yuvaltassa

HarshithaBR, 2024-03-21:
Thank you for your input on the error!
It helped resolve the issue we were facing.