google-deepmind/mujoco #1524: Implementation of 2-mass-system motor

SJH-JunhoSong, 2024-03-19:
Hi,

I'm a student and I'm trying to implement realistic actuator(torque) in MuJoCo.

To achieve this, I am comparing FRF results between a real plant and a MuJoCo model. 

This graph shows the FRF results where the input is torque and the output is angular velocity:

[image]

The blue line represents the FRF of the real plant, while the red line represents the MuJoCo actuator.

Unlike the blue line, which shows the characteristics of a 2-mass-system motor, the red line appears to resemble that of a one-mass motor.

I'm seeking assistance with implementing a 2-mass-system motor.

Here is a model which explains my question:
<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="RA_arm_link56">
    <compiler angle="radian" meshdir="assets/meshes/" />

    <option timestep="0.001" integrator="Euler" gravity = "0 0 -9.81">
      <flag sensornoise="enable" energy="enable" contact="enable" />
    </option>

    <include file="basic_scene.xml"/>

    <asset>
      <material name="dark" specular="0" shininess="0.25" rgba="0.8 0.8 0.8 1" />
      <!--<material name="dark"  rgba="0.2 0.2 0.2 1" />-->
      <material name="metal" rgba=".9 .95 .95 1"/>
      <material name="black" rgba="0 0 0 1"/>           
      <material name="white" rgba="1 1 1 1"/>
      <material name="gray"  rgba="0.671705 0.692426 0.774270 1"/>

        <mesh name="Link_base" file="Link_base_down.STL" />
        <mesh name="Link6" file="Link6_down.STL" />
    </asset>

    <worldbody>
		<body name="Link_base" pos="0 0 0.5" euler="1.57 -1.57 0">
			<inertial pos="0.0020312 -0.0045722 0.10227" euler="0 0 0" mass="2.109" diaginertia="0.00549 0.00487 0.00318"/>
			<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link_base" />

			<body name="Link6" pos="0.00191 0 0.118">
				<inertial pos="-8.80E-05 -0.0020431 0.12539" euler="0 0 0" mass="2.286" diaginertia="0.0088 0.008 0.00393" />
				<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link6" />
                <joint name="J6" type="hinge" pos="0 0 0" axis="0 1 0" stiffness="3700" armature="0.0846675" damping="0.00204"/>
			</body>
		</body>
	</worldbody>

    <actuator>
      <motor joint="J6" name="torque_J6" ctrllimited="true" ctrlrange="-50 50"/>
      <!--<position name="pos_J6" joint="J6" kp="0.1" />
      <velocity name="vel_J6" joint="J6" kv="0.1" />-->
    </actuator>

    <sensor>
      <jointpos joint="J6" name="ENC_J6" noise="0.0"/>
  		<!--<jointvel joint="J6" noise="0.0" />-->
  	</sensor>

	<!--<tendon>
	  <fixed name="tendon of joint">
		<joint joint="J6" coef="4000"/>
	  </fixed> damping="0.0043" armature="0.141" 
	</tendon>-->

  </mujoco>
```

</details>

yuvaltassa, 2024-03-19:
What is FRF?

What is a 2-mass-system motor?

SJH-JunhoSong, 2024-03-19:
Sorry for lack of explanation.

[image]

2-mass-system motor is the concept to represent movements of motor more realistically in engineering field.

It assumes that the change of motor angle isn't same with load angle, because of stiffness separating them.

So we use two equations as below.

[image]

Jl, \theta_L, Jm, \theta_m, \tau_m, K, N mean inertia of link, angle of link, inertia of motor, angle of motor, torque of motor, stiffness (Nm/rad), gear ratio, respectively. 

My question is: Is it possible to get the values of motor rotor's angle and the link angle separately? 

by the way, FRF means frequency response function, to obtain transfer function's characteristic in freq. domain.

yuvaltassa, 2024-03-19:
You want something like is described [here](https://mujoco.readthedocs.io/en/latest/modeling.html#backlash), but without the limits and armature. Basically just model the thing as it is: put two masses in series with large stiffness.

closed 2024-03-19 by yuvaltassa

SJH-JunhoSong, 2024-03-19:
> You want something like is described [here](https://mujoco.readthedocs.io/en/latest/modeling.html#backlash), but without the limits and armature. Basically just model the thing as it is: put two masses in series with large stiffness.

Is it ok to ask some examples for two masses in series with large stiffness?

yuvaltassa, 2024-03-19:
```xml
<body>
  <joint type="hinge" axis="0 0 1" />
  (geom or inertial, something that gives mass)
  <body>
    <joint type="hinge" axis="0 0 1" stiffness="100"/>
    (geom or inertial, something that gives mass)
  </body>
</body>

```

SJH-JunhoSong, 2024-03-19:
Thanks for kind help!

If I set stiffness="3700", dose it mena 3700 Nm/rad? (in my XML file, units are meter and radian respectively.)

yuvaltassa, 2024-03-19:
Yes

SJH-JunhoSong, 2024-03-19:
Is 'backlash' which use two joints on a single body available on version 2.2.1??

It seems backlash doesn't work on that release...

yuvaltassa, 2024-03-19:
Should be fine, what is not working for you?

SJH-JunhoSong, 2024-03-19:
I don't know why but when I added joint to implement motor rotor and stater, below the joint name j6, simulation stoped and didn't work.

Here is a my XML file.

<mujoco model="RA_arm_link56">
    <compiler angle="radian" meshdir="assets/meshes/" />

    <option timestep="0.001" integrator="Euler" gravity = "0 0 -9.81">
      <flag sensornoise="enable" energy="enable" contact="enable" />
    </option>

    <include file="basic_scene.xml"/>

    <asset>
      <material name="dark" specular="0" shininess="0.25" rgba="0.8 0.8 0.8 1" />
      material name="dark"  rgba="0.2 0.2 0.2 1" />-->
      <material name="metal" rgba=".9 .95 .95 1"/>
      <material name="black" rgba="0 0 0 1"/>           
      <material name="white" rgba="1 1 1 1"/>
      <material name="gray"  rgba="0.671705 0.692426 0.774270 1"/>

        <mesh name="Link_base" file="Link_base_down.STL" />
        <mesh name="Link6" file="Link6_down.STL" />
    </asset>

    <worldbody>
		<body name="Link_base" pos="0 0 0.5" euler="1.57 -1.57 0">
			<inertial pos="0.0020312 -0.0045722 0.10227" euler="0 0 0" mass="2.109" diaginertia="0.00549 0.00487 0.00318"/>
			<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link_base" />

			<body name="Link6" pos="0.00191 0 0.118">
				<inertial pos="-8.80E-05 -0.0020431 0.12539" euler="0 0 0" mass="2.286" diaginertia="0.0088 0.008 0.00393" />
				<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link6" />
                <joint name="J6" type="hinge" pos="0 0 0" axis="0 1 0" stiffness="3700" armature="0.0846675" damping="0.00204"/>
			</body>
		</body>
	</worldbody>

    <actuator>
      <motor joint="J6" name="torque_J6" ctrllimited="true"
      <material name="metal" rgba=".9 .95 .95 1"/>
      <material name="black" rgba="0 0 0 1"/>           
      <material name="white" rgba="1 1 1 1"/>
      <material name="gray"  rgba="0.671705 0.692426 0.774270 1"/>

        <mesh name="Link_base" file="Link_base_down.STL" />
        <mesh name="Link6" file="Link6_down.STL" />
    </asset>

    <worldbody>
		<body name="Link_base" pos="0 0 0.5" euler="1.57 -1.57 0">
			<inertial pos="0.0020312 -0.0045722 0.10227" euler="0 0 0" mass="2.109" diaginertia="0.00549 0.00487 0.00318"/>
			<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link_base" />

			<body name="Link6" pos="0.00191 0 0.118">
				<inertial pos="-8.80E-05 -0.0020431 0.12539" euler="0 0 0" mass="2.286" diaginertia="0.0088 0.008 0.00393" />
				<geom type="mesh" contype="1" conaffinity="0" group="1" material="dark" mesh="Link6" />
                <joint name="J6" type="hinge" pos="0 0 0" axis="0 1 0" stiffness="3700" armature="0.0846675" damping="0.00204"/>
			</body>
		</body>
	</worldbody>

    <actuator>
      <motor joint="J6" name="torque_J6" ctrllimited="true" ctrlrange="-50 50"/>
      
    </actuator>

    <sensor>
      <jointpos joint="J6" name="ENC_J6" noise="0.0"/>
  		
  	</sensor>

	

  </mujoco>