google-deepmind/mujoco #1531: Customizing the rendering range for individual cameras.

LinShuang22, 2024-03-20:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating cameras.

I'm looking for some help with customizing the rendering range for individual cameras. In my simulation environment, there are many cameras, and each camera requires a different rendering range. I know that I can adjust the camera's rendering range using `znear` and `zfar`. 

I have a few questions:

1. These parameters(`znear` , `zfar`) seem to be global. How can I set different rendering ranges for different cameras?

2. What can the `frustum` parameter be used for? I thought only objects within the frustum range could be rendered. I tried changing the size of the `frustum`, but the camera's rendered image doesn't seem to change. Objects outside the frustum are still visible in the camera. The rendering range seems to only be related to `znear` and `zfar`. 

Here is my test model:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <statistic extent="1"/>
  <visual>
    <map znear="0.01" zfar="10"/>
    <rgba frustum="1. 1. 0 0.2"/>
    <scale frustum="1"/>
  </visual>

  <asset>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".8 .6 .4" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="9 16" texuniform="false" reflectance=".2"/>
  </asset>

  <worldbody>
    <light pos="0 0 3"/>
    <geom type="plane" size="10 10 .01"/>
    <camera pos="0 2 1" xyaxes="0 -1 0 0 0 1" focal="8e-3 8e-3" resolution="1920 1200"
        sensorsize="9.6e-3 6e-3" principalpixel="200 0"/>
    <body pos="0 0 0">
        <joint type="slide" axis="1 0 0" limited="true" range="-1 20"/>
        <geom name="box2" type="box" size=".375 .6 .1" pos="1 2 1" material="grid" euler="0 90 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

[image]

the viewpoint of camera0.
[image]

aftersomemath, 2024-03-21:
Hi,

My understanding is that the `frustum` element of visual/scale is just for rendering a visualization of the camera's frustum. It does not actually effect the rendering process.

In theory you should be able to set [`MjvScene.mjvGLCamera.frustum_*`](https://mujoco.readthedocs.io/en/stable/APIreference/APItypes.html#mjvglcamera) as you see fit in order to set `znear` and `zfar` per camera.

These variables are normally computed from the global `znear` and `zfar` options during the call to [`mjv_updateCamera`](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mjv-updatecamera) which itself is called from [`mjv_updateScene`](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mjv-updatescene). Normally you would call `mjv_updateScene` before rendering a frame, however I think you should be able to recalculate `mjvGLCamera.frustum_*` between the calling `mjv_updateScene` and the subsequent call to `mjr_render`. I have not tried this myself though.

LinShuang22, 2024-03-29:
@aftersomemath Thank you, I also noticed the 'glcamera' low-level camera. But I don't know how to use it. Mujoco's description of it is a bit sparse. Could you provide an example program? : )

closed 2024-08-30 by kevinzakka