google-deepmind/mujoco #1536: Engineering units of body->geom->mass value

madhephaestus, 2024-03-21:
Hi there!

I have been looking through the XML reference https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-geom and under the `mass` term the documentation does not specify the units for that mass value. How are the units for this value determined? Is it documentation i am missing, or is a configuration? 

I have the same question of engineering units with https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator-general for 
`forcerange`

yuvaltassa, 2024-03-21:
Units are whatever you want them to be, as long as they are consistent and subject to some small caveats to do with the defaults. Read this https://mujoco.readthedocs.io/en/latest/overview.html#units-are-unspecified

closed 2024-03-21 by yuvaltassa

madhephaestus, 2024-03-22:
OK, thank you! It might be helpful to add that link to each field in the XML documentation where the units are inferred like that, and even better if the MKS default units were listed. It is super cool that the units are not fixed, but sometimes it could be helpful to document the defaults, especially since the default gravity vector sets MKS everywhere *unless* the user changes it. that means the rest of the system defaults to a units, so it is easiest for developers to know the defaults to use them.