google-deepmind/mujoco #1540: Remove dynamics from simulation

aleada23, 2024-03-22:
### Discussed in https://github.com/google-deepmind/mujoco/discussions/1533

<div type='discussions-op-text'>

<sup>Originally posted by **aleada23** March 21, 2024</sup>
Good morning. I'm a student and I'm using MuJoCo for a project. I know that dynamic is an important feature for this engine, but I need to remove it from my environment. 
I've created an environment, also using Robosuite, where a robot has to perform tasks. Now, I need to remove the dynamics from the environment to make computation lighter for implementing a new learning component. Is it possible to remove the dynamics? I would like to avoid any calculations related to it, as it isn't necessary in this new phase. The dynamics currently does not constitute a fundamental part of my simulation during the learning phase, so I can do without it. I am only interested in keeping some functionalities active, such as the rangefinder sensors, or obtaining the Jocobian for solving inverse kinematics for robot control. Therefore, I would like to do away with the dynamics completely removing the values. For instance, if I were to zero out masses and frictions, I would remove a significant portion of the dynamics from the simulation, but I believe the calculations would still be performed, keeping the simulation slow as it is now.
Any suggestion about this?
Thank you for your time</div>

Balint-H, 2024-03-23:
Yuval's suggestion under your previous post would probably work for you. If you have your own way of updating or setting qpos/qvel, you can just call individual sub-steps from mj_step as needed.

I recommend reading up of the simulation loop section in the [docs](https://mujoco.readthedocs.io/en/stable/programming/simulation.html#simulation-loop), especially the breakdown of `mj_step1`. you can check the source code of the substeps inside `mj_step1` (e.g. for [mj_fwdPosition](https://github.com/google-deepmind/mujoco/blob/2019d42ca67d9eb7e3bb01eacb97ad98f9d0d4f7/src/engine/engine_forward.c#L123)) or their summaries in the [API documentation](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html), if you need to go even more fine grained.

aleada23, 2024-03-26:
Thank you, I'll check the documentation that you provided.

closed 2024-08-15 by kevinzakka