google-deepmind/mujoco #1545: contact problems in MJX

gautica, 2024-03-25:
Hey,
I found there is some contact problems in MJX. I am using PD control to make quadrupeds stand at default joint positions. It works fine in Mujoco, but with the same setups (stiffness, damping) it doesn't work in MJX. I rendered the contact force, and it seems the contact force is changing kind of randomly, even the robot just stands there. Here is the video of it. The robot should just stand there. Thank you for any help.

https://github.com/google-deepmind/mujoco/assets/33986182/8c40165e-ee44-4d67-be30-f52ec325121c

btaba, 2024-03-26:
Hi @gautica , can you send us an XML so we can debug? Likely you'll need to tune the solver params, see https://mujoco.readthedocs.io/en/stable/mjx.html#performance-tuning

btaba, 2024-05-23:
There's an MJX version of the go2, so closing this issue. PTAL at https://github.com/google-deepmind/mujoco_menagerie/tree/main/unitree_go2

closed 2024-05-23 by btaba