google-deepmind/mujoco #1548: Help Troubleshooting Joint Display After Rajagopal2016.osim to MuJoCo Conversion

AliForo, 2024-03-26:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for my PhD thesis. I'm new to MuJoCo and attempting to convert the Rajagopal_2016 OpenSim model to a MuJoCo XML file. I'm using the myoconverter tool, and the conversion process completes successfully. However, when I load the converted model in MuJoCo, the joints are not rendering as expected.
I'm not sure why my joints aren't rendering correctly after conversion. If anyone has experience with similar issues, what troubleshooting steps would you recommend?

I've attached a PDF of the XML file (original file was too large), a screenshot, and a log file to help troubleshoot the issue we discussed.

Thanks!
[xlm.pdf](https://github.com/google-deepmind/mujoco/files/14761739/xlm.pdf)
[image]
[Rajagopal2016_conversion.log](https://github.com/google-deepmind/mujoco/files/14761605/Rajagopal2016_conversion.log)

Vittorio-Caggiano, 2024-03-26:
MyoConverter provides a conversion of the opensim models which is optimized with respect to the original information but it is not directly optimized to be robust to simulations performed in MuJoCo. You can look at the [MyoLeg](https://github.com/MyoHub/myo_sim/tree/main/leg) (which was based on the Rajagopal) which we have developed (starting from the myoconverter) and optimized to be robust to simulations with [MyoSuite ](https://github.com/MyoHub/myosuite)

Ali-Foro, 2024-03-26:
Thank you for your insights on MyoConverter and the MyoLeg model. I'm working with experimental walking data and need to convert the scaled Rajagopal model to MuJoCo format. I previously managed to convert the scale gait2354 model successfully, but I'm facing challenges with the Rajagopal model, which isn't functioning as expected in MuJoCo. Can you provide guidance on how to adapt or optimize the Rajagopal model for MuJoCo, similar to what was achieved with the MyoLeg model? Specific steps or considerations to ensure a successful conversion and simulation in MuJoCo would be very helpful.
Additionally, could you guide me on how to obtain the osim file of the MyoLeg model scaled to my specifications, and then convert it to MuJoCo format? Detailed steps or resources to acquire, scale, and convert the MyoLeg model for MuJoCo simulations would be invaluable.