google-deepmind/mujoco #1550: The effect of impedance on different geometries

yamanyigit, 2024-03-26:
I am currently pursuing a master's degree, and my research is centered at the soft robotics. Our work involves modeling robots using Mujoco. Through our experiments, I have observed that while a sphere interacting with the floor is modeled with a single contact point, a box in similar circumstances is represented by four contact points. Notably, when a sphere penetrates the floor, decreasing the impedance—constant across all constraint violations—results in a reduced force exerted on the sphere, aligning with the Mujoco documentation's expectations.

However, an intriguing discrepancy arises with a box under similar conditions. Upon decreasing the impedance, the force exerted on the box unexpectedly increases, suggesting an escalation in spring and damping coefficients contrary to the decrease in impedance. This outcome is inconsistent with the behavior observed with the sphere and contradicts the guidelines provided in the Mujoco documentation. I am perplexed by this apparent contradiction, as my understanding leads me to expect a decrease in force analogous to the sphere case when impedance is reduced.

Can you please provide insight into this phenomenon or suggest potential reasons for this unexpected behavior?

Balint-H, 2024-03-27:
I did a test using the solver override options of simulate. To illustrate, here are two screenshots of a paused simulation changing the maximum value of d :

[image]
[image]

Where have you noticed this effect? Only on the way the contact forces are rendered, or in the actual values as well? Unfortunately I don't have an answer for you, but hopefully these images help give context.

yamanyigit, 2024-03-27:
[image]
[image]

I have noticed this effect on both cases: the rendered and the actual values