google-deepmind/mujoco #1553: How to use weld to fix a cable on the link?

Wangzihao12ous, 2024-03-27:
I used weld to fix a cable on the link2 , but in simulate if i  apply force on the cable ,the cable will be separated from the link2.

Here is the code:
        <weld name="right_boundary0" body1="0B_last" body2="link2_cable_1" anchor="0 0 0" />
        <weld name="right_boundary1" body1="1B_last" body2="link2_cable_2" anchor="0 0 0" />
        <weld name="right_boundary2" body1="2B_last" body2="link2_cable_3" anchor="0 0 0" />

Wangzihao12ous, 2024-03-27:
I used weld to fix a cable on the link2 , but in simulate if i  apply force on the cable ,the cable will be separated from the link2.

Here is the code:
        <weld name="right_boundary0" body1="0B_last" body2="link2_cable_1" anchor="0 0 0" />
        <weld name="right_boundary1" body1="1B_last" body2="link2_cable_2" anchor="0 0 0" />
        <weld name="right_boundary2" body1="2B_last" body2="link2_cable_3" anchor="0 0 0" />

The gray ball is the weld point.
[image]

Wangzihao12ous, 2024-03-27:
And how to make the cable through the small hole in link?
The cable have no gemo name,so i cant use contact pair.
[image]

yuvaltassa, 2024-04-11:
why are you not using tendons?

Wangzihao12ous, 2024-04-12:
> why are you not using tendons?

Thank you! I need to use rigid cable in the simulation，ensure that the force on the cable is consistent and there is as little deformation as possible.

yuvaltassa, 2024-04-15:
And which part of this do you think you cannot do with a tendon?

Wangzihao12ous, 2024-04-16:
In the model I hope to achieve, I can change the position of the first site where the tendon is located, in order to change the length of the tendon. I used muscles and general as actuator, but the results were not very good.

Wangzihao12ous, 2024-04-16:
> And which part of this do you think you cannot do with a tendon?

In the model I hope to achieve, I can change the position of the first site where the tendon is located, in order to change the length of the tendon. I used muscles and general as actuator, but the results were not very good.

Wangzihao12ous, 2024-04-16:
> And which part of this do you think you cannot do with a tendon?

Thank you!In the model I hope to achieve, I can change the position of the first site where the tendon is located, in order to change the length of the tendon. I used muscles and general as actuator, but the results were not very good.