google-deepmind/mujoco #1554: force sensor with connect

2-woo-10, 2024-03-27:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

I'm a student and I'm trying to use MuJoCo for estimation about contact force.
I'm looking for some help.

I'm using a force sensor to measure the force by declaring a gem at the end of the franka panda gripper.
A total of 4 geoms are made on one side of the gripper to measure the force, and after checking the values, a vibrating shape appears, can you explain this?
<details>
<mujoco model="panda">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>
  <option integrator="implicit" timestep="0.001"/>
  <option gravity="0 0 -9.81"/>
  <option>
  <flag multiccd="enable"/>
  </option>
  <default>
    <default class="panda">
      <material specular="0.5" shininess="0.25"/>
      <joint frictionloss="2" damping="1" axis="0 0 1" range="-2.7437 2.7437"/>

      
      <default class="finger">
        <joint axis="0 1 0" type="slide" range="0 0.04"/>
      </default>

      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="collision">
        <geom type="mesh" group="3"/>
        <default class="fingertip_pad_collision_1">
          <geom type="box" size="0.0085 0.004 0.0085" pos="0 0.0055 0.0445"/>
        </default>
        <default class="fingertip_pad_collision_2">
          <geom type="box" size="0.003 0.002 0.003" pos="0.0055 0.002 0.05"/>
        </default>
        <default class="fingertip_pad_collision_3">
          <geom type="box" size="0.003 0.002 0.003" pos="-0.0055 0.002 0.05"/>
        </default>
        <default class="fingertip_pad_collision_4">
          <geom type="box" size="0.003 0.002 0.0035" pos="0.0055 0.002 0.0395"/>
        </default>
        <default class="fingertip_pad_collision_5">
          <geom type="box" size="0.003 0.002 0.0035" pos="-0.0055 0.002 0.0395"/>
        </default>
      </default>
    </default>
  </default>

  <asset>
    <texture name="grid" builtin="checker" height="500" rgb1=".1 .2 .3" rgb2=".2 .3 .4" type="2d" width="500"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>

    <material class="panda" name="white" rgba="1 1 1 1"/>
    <material class="panda" name="off_white" rgba="0.901961 0.921569 0.929412 1"/>
    <material class="panda" name="black" rgba="0.25 0.25 0.25 1"/>
    <material class="panda" name="green" rgba="0 1 0 1"/>
    <material class="panda" name="light_blue" rgba="0.039216 0.541176 0.780392 1"/>

    
    <mesh name="link0_c" file="link0.stl"/>
    <mesh name="link1_c" file="link1.stl"/>
    <mesh name="link2_c" file="link2.stl"/>
    <mesh name="link3_c" file="link3.stl"/>
    <mesh name="link4_c" file="link4.stl"/>
    <mesh name="link5_c0" file="link5_collision_0.obj"/>
[... 133 lines cut ...]

  <sensor>
    <force name="left_dummy_site_1" site="left_dummy_site_1"/>
    <force name="left_dummy_site_2" site="left_dummy_site_2"/>
    <force name="left_dummy_site_3" site="left_dummy_site_3"/>
    <force name="left_dummy_site_4" site="left_dummy_site_4"/>

    <force name="right_dummy_site_1" site="right_dummy_site_1"/>    

    <force name="right_dummy_site_2" site="right_dummy_site_2"/>
    <force name="right_dummy_site_3" site="right_dummy_site_3"/>
    <force name="right_dummy_site_4" site="right_dummy_site_4"/>
    
    

  </sensor>

</mujoco>

</details>

Here is a screenshot / video, illustrating my question:
[image]
[Screencast from 2024년 03월 27일 22시 33분 31초.webm](https://github.com/google-deepmind/mujoco/assets/109202740/441d7bad-1141-408b-926e-6e505a33c8fe)
09202740/6cd0a2a3-eb1d-4e4e-88fc-00c6cc35d6d6)

yuvaltassa, 2024-03-27:
This is common and not unphysical. Force-vibration due to fast stick-slip is something that happens.

Try reducing the timestep and see what effect that has.

2-woo-10, 2024-03-27:
This is the result of setting the current timestep to 0.05.
[Screencast from 2024년 03월 27일 23시 18분 06초.webm](https://github.com/google-deepmind/mujoco/assets/109202740/aeae2a9d-6d00-4a7d-ab04-bb11cfe984a8)

2-woo-10, 2024-03-27:
@yuvaltassa

This is the result of setting the current timestep to 0.05.
[Screencast from 2024년 03월 27일 23시 18분 06초.webm](https://github.com/google-deepmind/mujoco/assets/109202740/aeae2a9d-6d00-4a7d-ab04-bb11cfe984a8)

yuvaltassa, 2024-04-22:
50ms is huge! Try 1ms

closed 2024-08-30 by kevinzakka