google-deepmind/mujoco #1556: Contact between Panda hand and Flex object

songsq21, 2024-03-27:
Hi,

I'm an undergraduate student and I'm trying to use MuJoCo to simulate on cloth manipulation.

I'm looking for some help with the collision between Franka Panda hand and Flex hand.

Here is my question.

### My Simulation Setting

I intend to do some experiment on cloth manipulation, where I used Franka Panda to grip the cloth, which is in fact a **flexcomp**
 object, which is the advice from @quagla. My Panda hand is from the menagerie repo and my flexcomp object is defined in the following XML file.

<details>
  <summary>flexcomp XML as cloth</summary>

```XML

<mujoco>
    <size memory="1024M"/>
    <include file="scene.xml"/>
    <include file="panda.xml"/>
    <extension>
        <plugin plugin="mujoco.elasticity.shell"/>
        <plugin plugin="mujoco.elasticity.membrane"/>
    </extension>
    <option timestep="0.001" solver="CG" tolerance="1e-6" integrator="implicitfast"/>
    <visual>
        <map stiffness="100"/>
    </visual>
    <!-- Asset Cloth by flex -->
    <worldbody>
        <flexcomp type="grid" count="9 9 1" spacing=".03 .03 .03" pos="0.4 0.4 0"
                radius=".002" mass="1" name="plate" dim="2">
        <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001"/>
        <edge equality="false" damping="10"/>
        <pin id="54 72"/>
        <plugin plugin="mujoco.elasticity.membrane">
            <config key="poisson" value="0"/>
            <config key="thickness" value="1e-2"/>
            <!-- Units are in Pa (SI) -->
            <config key="young" value="3e5"/>
        </plugin>
        </flexcomp>
    </worldbody>
</mujoco>

```
</details>

### My Questions Encountered

In my experiment, I found out that I couldn't grab the flexcomp object well with the hand:
* The first problem is that the hand cannot seize the cloth tightly, as space between the hand and the hand can be seen even if I set the actuator for the fingers in the hand to the tightest.
* The second problem is that the cloth often slips off from the hand, which also happens when I tried to grab other objects like spheres and cubes. I could change the friction settings for those objects to alleviate the slippery, but not for the flex object. 

Here is a video illustrating my question:

https://github.com/google-deepmind/mujoco/assets/129511833/c9427c96-55fc-4ca9-8a2b-365de14e57e5

dblanm, 2024-05-15:
This seems as the same issue in #1480

kevinzakka, 2024-08-30:
Closing as duplicate of #1480.

closed 2024-08-30 by kevinzakka