google-deepmind/mujoco #1560: Unexpected force when impedance is changed for a box geometry

yamanyigit, 2024-03-29:
I am currently pursuing a master's degree, and my research is centered at the soft robotics. Our work involves modeling robots using Mujoco. Through our experiments, I have observed that while a sphere interacting with the floor is modeled with a single contact point, a box in similar circumstances is represented by four contact points. Notably, when a sphere penetrates the floor, decreasing the impedance—constant across all constraint violations—results in a reduced force exerted on the sphere, aligning with the Mujoco documentation's expectations.

However, an intriguing discrepancy arises with a box under similar conditions. Upon decreasing the impedance, the force exerted on the box unexpectedly increases, suggesting an escalation in spring and damping coefficients contrary to the decrease in impedance. This outcome is inconsistent with the behavior observed with the sphere and contradicts the guidelines provided in the Mujoco documentation. I am surprised by this apparent contradiction, as my understanding leads me to expect a decrease in force analogous to the sphere case when impedance is reduced.

I think this unexpected behavior is an error.

Impedance values do not change with the increase in violation. They are constant.

Sphere Case:

Impedance: 0.9 0.9 0.001
[image]

Impedance: 0.3 0.3 0.001
[image]

Box Case:

Impedance: 0.9 0.9 0.001
[image]

Impedance: 0.3 0.3 0.001
[image]

  Operating system: Windows 11
  MuJoCo version: 3.1.3

yuvaltassa, 2024-04-11:
This is indeed quite odd. Thanks for bringing it to our attention. I will look more carefully and get back to you.

yuvaltassa, 2024-04-26:
Hi @yamanyigit, finally getting around to this :slightly_smiling_face: 

I've tried and failed to reproduce what you've reported. I do believe you've seen something real, so perhaps my attempt at reproduction is wrong. See my model and results below. Reducing `solimp[0, 1]` (using `override`) is reducing the contact forces for both box and sphere, as expected.

<details>
  <summary>debugging XML</summary>

```XML
<mujoco model="debug">
  <option gravity="0 0 -10" o_solimp=".3 .3 1e-3"/>

  <visual>
    <map force="0.05"/>
  </visual>

  <default>
    <geom solimp=".9 .9 1e-3"/>
  </default>

  <worldbody>
    <light pos=".1 0 .6"/>
    <light pos="-.1 0 .6"/>
    <geom type="plane" size="1 1 .01"/>

    <body pos=".12 0 .1" >
      <freejoint/>
      <geom mass="10" type="sphere" size="0.1" rgba=".8 0 0 1"/>
    </body>

    <body pos="-.12 0 .1" >
      <freejoint/>
      <geom mass="10" type="box" size=".1 .1 .1" rgba="0 0 .8 1"/>
    </body>
  </worldbody>
</mujoco>
```

</details> 

[image]
[image]

yaman-y, 2024-05-08:
Thank you for your response. However, I release the objects from a certain height and then I pause the simulation after the objects slightly penetrate into the ground. In other words, the objects are not in equilibrium. The screenshots I shared were taken slightly after the objects hit the ground. Your screenshots are taken at rest.

yuvaltassa, 2026-09-14:
Closing due to lack of activity, please open a new issue if this is still a problem

closed 2026-09-14 by yuvaltassa