google-deepmind/mujoco #1567: Correct use of intvelocity actuator for controlling velocity and position

MiaoDragon, 2024-04-03:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for my research on object pushing through MPC, especially using mujoco-mpc.

I'm looking for some help with the intvelocity actuator. I have used this for controlling the joint velocity and position of a motoman robot in another project, where I defined intvelocity actuator for each joint, and tracks a given velocity and position command (e.g. generated by spline interpolation) through:

```
data.act[intact] = position
data.ctrl[intvctrl] = velocity
```
This is working well for tracking both position and velocity command (actually better than combining velocity and position actuators). However, I question whether this is the right way to apply actuation. It seems that the `act` field is part of the state, rather than control. I face trouble using it with mujoco-mpc, where control is only applied from `ctrl` field, rather than `act`. At the same time, when not updating the `act` field, it seems my 

Hence I'd like to ask if there would be any clarification or help on how to use intvelocity to control position and velocity. Or could anyone recommend any correct way to control position and velocity through mujoco actuators? I would really appreciate your help!

One example XML file I have is the following:
```
<?xml version="1.0" encoding="utf-8"?>
<mujoco model="scene1">
  <asset>
    <texture builtin="checker" height="512" name="grid" rgb1="0.1 0.2 0.3" rgb2="0.2 0.3 0.4" type="2d" width="512"/>
    <material name="grid" emission="1" reflectance="0" shininess="0" specular="0" texrepeat="0.5 0.5" texture="grid" texuniform="true"/>
  </asset>

  <worldbody>
    <geom condim="3" material="grid" name="floor" size="2 2 0.05" type="plane"/>
    <geom conaffinity="0" contype="0" group="1" material="grid" name="floor_vis" size="2 2 0.05" type="plane"/>
    <light castshadow="false" diffuse="1 1 1" dir="0 0 -1" directional="true" pos="-0.5 0.5 3"/>

    <body name="ee_position" pos="0 0 0.1" gravcomp="1">
      <camera name="cam0" pos="0 -0.3 0.3" xyaxes="1 0 0 0 0.7 0.7"/>
      <joint name="ee_position_x" type="slide"  pos="0 0 0" axis="1 0 0" ref="0" />
      <joint name="ee_position_y" type="slide"  pos="0 0 0" axis="0 1 0" ref="0" />
      <joint name="ee_position_z" type="slide"  pos="0 0 0" axis="0 0 1" ref="0.1" />
      <geom name="ee_position" type="sphere" size=".005" mass="10"/>
      <site name="tip" pos="0 0 0" size="0.001"/>
    </body>
  </worldbody>

  <actuator>

    <intvelocity name="ee_position_x_vel" joint="ee_position_x" kp="1000" kv="100" actrange="-2 2" ctrllimited="true" ctrlrange="-0.1 0.1"/>
    <intvelocity name="ee_position_y_vel" joint="ee_position_y" kp="1000" kv="100" actrange="-2 2" ctrllimited="true" ctrlrange="-0.1 0.1"/>
    <intvelocity name="ee_position_z_vel" joint="ee_position_z" kp="1000" kv="100" actrange="0 2" ctrllimited="true" ctrlrange="-0.1 0.1"/>

  </actuator>
</mujoco>
```

The following video might illustrate my problem a bit. Not sure if you can see it clearly, but the ball at beginning moved very fast to the ground. This is because I used `mj_forward()` to move the ball position to another place, but didn't change the `act` field. So I assume the `act` field kind of controls the position part. Is my assumption correct?
https://github.com/google-deepmind/mujoco/assets/15926481/c05d1b51-b5e9-4fbc-bf98-61b274e20ad3

If you could suggest the right way to use `intvelocity` I would really appreciate it. Thanks!

yuvaltassa, 2024-04-04:
You shouldn't be writing into `act`, it is indeed a part of the state.

yuvaltassa, 2024-04-05:
Most PID controllers are really just P controllers with a damping term. This is just a `position` controller with nonzero `kv`.

A "proper" PID controller is implemented [as a plugin](https://github.com/google-deepmind/mujoco/tree/main/plugin/actuator#readme)

closed 2024-04-05 by yuvaltassa

ryznir, 2025-04-07:
If I wanted to control a hinge joint so that it moves at say 1 rad/s and compensates torque when it hits resistance to remain at 1 rad/s, would the PID plugin be my best bet? I've been having issues with the intvelocity and velocity actuators, they are also vague on what should be input to them