google-deepmind/mujoco #1569: How to use urdf files in mujoco?

LIUHAITAO-CH, 2024-04-04:
macOS Sonoma 14.4.1

Mujoco 3.1.3(`pip install mujoco`，not C++)

How to use urdf files in mujoco? , should the urdf file be modified appropriately, or does mujoco have an API to directly process urdf files? And what is the general format of the mesh path in the urdf file?

Thanks

LIUHAITAO-CH, 2024-04-04:
and i noticed that the [URDF Extensions](https://mujoco.readthedocs.io/en/stable/modeling.html#curdf), this picture is my files
[image]
What is the correct way to write **meshdir** and **mesh**, and what is needed to correctly identify and compile this urdf file in mujoco?
[image]
 thanks

jonzamora, 2024-05-10:
@LIUHAITAO-CH I see you are using the Franka Panda arm. I'd recommend you check out the [Franka Panda arm from MuJoCo Menagerie](https://github.com/google-deepmind/mujoco_menagerie/tree/main/franka_emika_panda) if your main goal is to have a MuJoCo environment w/ the Panda arm already setup for you.

LIUHAITAO-CH, 2024-05-13:
> @LIUHAITAO-CH I see you are using the Franka Panda arm. I'd recommend you check out the [Franka Panda arm from MuJoCo [...]

thanks for your suggest, Franka is just an example of my test. What I want to realize is a mobile manipulator model, which is a 4-wheel differential base and franka as a whole.

traversaro, 2024-05-13:
> How to use urdf files in mujoco? , should the urdf file be modified appropriately, or does mujoco have an API to directly process urdf files? And what is the general format of the mesh path in the urdf file?

Just in case someone finds this discussion with search engines, there have been a discussion on this in https://github.com/google-deepmind/mujoco/issues/1432 . Main points I got from the discussion (if I write anything wrong feel free to correct me):
* MuJoCo supports loading URDF, but does not support out of the box the use of `package://` URIs, that is how URDF specification prescribe mesh path should be specified. Instead, as you correctly pointed out, relative paths (not directly supported in URDF spec) should be used for meshes, and `meshdir` should be used to indicate the relative location of the mesh directory w.r.t. to the `.urdf` file directory. In https://github.com/google-deepmind/mujoco/issues/1569#issuecomment-2036673558  example so `meshdir` should also include the `meshes` directory, or the `strippath` should be set to `false` (it is `true` by default for URDF files.
* Beside meshes related modifications, it is common for people consuming URDF in mujoco to modify the URDF before passing to mujoco, and then eventually also modify the resulting `mjcf` file (see for example https://github.com/google-deepmind/mujoco/issues/273#issuecomment-1124060013), as some features are not available in the form of URDF extensions.

yuvaltassa, 2024-05-14:
Thanks @traversaro !

How hard/easy would it be to add URI support? The URDF import code is in [xml_urdf.cc](https://github.com/google-deepmind/mujoco/blob/main/src/xml/xml_urdf.cc), which I think is not very hard to read. Let us know if there are straightforward improvement that you'd like us to add. Better still, send us a PR 🙂

closed 2024-05-14 by yuvaltassa

traversaro, 2024-05-14:
> How hard/easy would it be to add URI support? The URDF import code is in [xml_urdf.cc](https://github.com/google-deepmind/mujoco/blob/main/src/xml/xml_urdf.cc), which I think is not very hard to [...]

I did a working prototype in https://github.com/google-deepmind/mujoco/issues/1432#issuecomment-1962609851, however I realized that anyhow this can't be used to load out of the box URDF with `package://` URIs, as you need anyhow to:
* Modify the URDF to set `strippath` to `false`
* Modify the C++/Python source code to register the resource provider

So we pivoted to have code to modify the URDF files on the fly, as that is necessary for other parameters that can't be added as URDF extensions. See https://github.com/google-deepmind/mujoco/issues/1432#issuecomment-1962918624 for a more in-depth discussion.

LIUHAITAO-CH, 2024-05-14:
thanks, @traversaro @yuvaltassa .now i have another question about mujoco visulaization.
in urdf, my code is shown below. 
```
<?xml version="1.0" ?>

<robot name="fr3">
  
  <mujoco>
    <compiler meshdir="meshes/" balanceinertia="true" discardvisual="false"/>
  </mujoco>

  <link name="fr3_link0">
    <inertial>
      <origin xyz="-0.041018 -0.00014 0.049974" rpy="0 0 0" />
      <mass value="0.629769" />
      <inertia ixx="0.00315" ixy="8.2904E-07" ixz="0.00015" iyy="0.00388" iyz="8.2299E-06" izz="0.004285" />
    </inertial>

    <visual>
      <geometry>
        <mesh filename="link0_meshlab.stl"/>
      </geometry>
    </visual>

    <collision>
      <geometry>
        <mesh filename="link0.stl"/>
      </geometry>
    </collision>
  </link>

  <link name="fr3_link1">
    <inertial>
      <origin xyz="0.003875 0.002081 -0.04762" rpy="0 0 0" />
      <mass value="4.970684" />
      <inertia ixx="0.70337" ixy="-0.000139" ixz="0.006772" iyy="0.70661" iyz="0.019169" izz="0.009117" />
    </inertial>

    <visual>
      <geometry>
        <mesh filename="link1_meshlab.stl"/>
      </geometry>
    </visual>

    <collision>
      <geometry>
        <mesh filename="link1.stl"/>
      </geometry>
    </collision>
  </link>

......

```
[image]
[image]

I found that mujoco is not displaying a visual pattern, but a convex hull model in collision, and if the collision label is commented out, the normal visual label model will be displayed. Why is this?
[image]

yuvaltassa, 2024-05-14:
Did you read this?

https://mujoco.readthedocs.io/en/latest/modeling.html#urdf-extensions

https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-discardvisual

LIUHAITAO-CH, 2024-05-14:
@yuvaltassa hello, I did read the [document](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-discardvisual), i think that `discardvisual` is `true` for urdf, but the result is wrong,still a convex hull.

yuvaltassa, 2024-05-14:
I think you are turning on the convex hull visualization. Try hitting the "H" key for both models, see what happens.

yuvaltassa, 2024-05-14:
try pressing the 0, 1 or 2 keys, I think the visual and collision geoms are usually in different groups

LIUHAITAO-CH, 2024-05-15:
@yuvaltassa thanks,you are right,
group 0 is enabled by default, when i press the 0 key ， the convex hull doesn't show up
[image]

when i press the 0 key again,  the convex hull show up
[image]

Can I not set it in the mujoco tag of the urdf file so that it does not show packages? Or is it only possible to convert urdf files to mjcf files?

help me，thanks