google-deepmind/mujoco #1576: How to create joints between boies/joints and change the joint status dynamically

dibbla, 2024-04-07:
Hi,

I am a student working on a project, where I have multiple balls which can roll around freely. One feature I would like to add is some pin-joints/constraints between balls. When these constraints are active, 2 balls are connected together and cannot separate, but balls can roll-over/climb-over the other connected balls.

To better illustrate my idea, I draw a picture:
[image]

In addition, I would like to switch between active/inactive under some conditions (e.g. when they collide) when running the simulation. It is a bit like PinJoint in PyMunk simulation.

I have get a plain implementation with balls, but I cannot find similar implementations:

```python
import mujoco
import mujoco_viewer

# set up the ground and light and balls
xml="""
<mujoco>
    <option gravity="0 0 -9.81" />
    <worldbody>
        <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
        <geom type="plane" size="2 2 1" rgba=".8 .8 .8 1"/>
        <body pos="0 0 .5" name="ball_1">
            <joint name="ball_1" type="free"/>
            <geom type="sphere" size=".1" rgba=".5 .9 .9 1"/>
        </body>
        <body pos="0 .2 .1" name="ball_2">
            <joint name="ball_2" type="free"/>
            <geom type="sphere" size=".1" rgba=".5 .9 .9 1"/>
        </body>
        <body pos="0 .4 .1" name="ball_3">
            <joint name="ball_3" type="free"/>
            <geom type="sphere" size=".1" rgba=".5 .9 .9 1"/>
        </body>
    </worldbody>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

# create the viewer object
viewer = mujoco_viewer.MujocoViewer(model, data)

# simulate and render
for _ in range(10000):
    if viewer.is_alive:
        mujoco.mj_step(model, data)
        viewer.render()
    else:
        break

    # sample spme random position from 0 to 1
    pos = [0.005, -0.005, 0.]
    # set the velocity of the ball_3
    print("ball_3:", data.joint('ball_3').qpos)
    data.joint('ball_3').qvel[:3] += pos
    mujoco.mj_step1(model, data)

    # get the collision information
    collision = data.ncon
    print("collision:", collision)
    for i in range(collision):
        print("contact:", data.contact[i].geom1, data.contact[i].geom2)

# close
viewer.close()
```

Any idea of implementing this would be appreciated!

yuvaltassa, 2024-04-07:
You want to use equality constraints for this, not joints. These can be in/activated at run time.

closed 2024-04-07 by yuvaltassa

dibbla, 2024-04-21:
Hi @yuvaltassa ! Thank you for the help. I have now implmented a minimal demo using the equality constraints. However, I still have some questions regrading the equality constraint.

I have defined the sphere radius to be 0.06, and I set the equality constraint as `<connect body1="ball_1" body2="ball_2" anchor="0.06 0 0.06" active="true"/>`. However, it seems like the rotation process is not as expected.

The xml file is:

```xml
<mujoco>
  <option gravity="0 0 0"/>

  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3"
    rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="2 2" texuniform="true"
    reflectance=".2"/>
  </asset>

  <worldbody>
    <light pos="0 0 2" mode="trackcom"/>
    <geom name="ground" type="plane" pos="0 0 -.5" size="2 2 .1" material="grid" solimp=".99 .99 .01" solref=".001 1" friction="1"/>
    <camera name="y" pos="0.06 -1.2 0" xyaxes="1 0 0 0 1 2"/>

    <body name="ball_1" pos="0 0 -0.44">
      <freejoint name="ball_1_joint"/>
      <geom name="ball_1_geom" size=".06" pos="0 0 0" rgba="1 0 0 1"/>
    </body>

    <body name="ball_2" pos="0.12 0 -0.44">
      <freejoint name="ball_2_joint"/>
      <geom name="ball_2_geom" size=".06" pos="0 0 0" rgba="0 1 0 1"/>
    </body>

  </worldbody>

  <equality>
    <connect body1="ball_1" body2="ball_2" anchor="0.06 0 0.06" active="true"/>
  </equality>

</mujoco>
```

And my goal is to rotate the ball around the other. I achieve this by setting the velocity of `ball_1`:

```python
# red is ball 1
# green is ball 2

n_frames = 300
height = 300
width = 600
frames = []

# load
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model, height, width)

# simulate and render
mujoco.mj_resetData(model, data)
for i in range(n_frames):
    while data.time < i/30.0:
        # get position information of all balls
        pos1 = data.geom("ball_1_geom").xpos
        pos2 = data.geom("ball_2_geom").xpos

        direction_vector = pos1 - pos2
        unit_direction = direction_vector / np.linalg.norm(direction_vector)

        radius = 0.06
        speed = -0.1
        perpendicular_vector = np.array([0, 1,0])
        velocity = speed * np.cross(unit_direction, perpendicular_vector)

        data.qvel[0:3] = velocity

        mujoco.mj_step(model, data)
        # data.qvel[1] = 0.1
    renderer.update_scene(data, "y")
    frame = renderer.render()
    frames.append(frame)
media.show_video(frames, fps=30)
```

The `ball_1` is designed to rotate around the `ball_2` while their relative distance is constrained. When I set the equlity constraint to be `false`, it seems fine but the relative distance constraint is not applied.

https://github.com/google-deepmind/mujoco/assets/72083122/4bca8799-654f-4440-a6e5-a206c0a3881e

However, when I activate the equality constraint, there is an obvious "rise" of the `ball_2`:

https://github.com/google-deepmind/mujoco/assets/72083122/7fba45d6-8435-4480-8efe-007241b54c5b

The same thing is observed if I change the constraint to be weld `<weld body1="ball_1" body2="ball_2" relpose="0.06 0 0.06 0 0 0 0" anchor="0.06 0 .06" active="true"/>`. 

My ultimate goal is to have the relative distance fixed while activate/deactivate in the air (I cannot use the euqality/joint since the balls are free joints). Do you have any suggestions/ideas about implementing this? Thanks a lot!