google-deepmind/mujoco #1577: Collisions in urdf-models

santerioksanen, 2024-04-08:
Hello,

I am working on a school project, migrating a MuJoCo <-> ROS interface from MuJoCo 2 to MuJoCo 3. I am having issues with collision calculation based on a `urdf` defined model, which was working fine with MuJoCo 2. With the simple example below, a ball drops right through the floor.

Interestingly if I change the joint type between floor and base_link to anything but fixed, collisions are applied. Furthermore, if I compile the urdf to xml, and again the xml to the binary file, collisions are applied.

This seems to be related to the parsed `body_contype` and `body_conaffinity` being `0` for the floor, while they are `1` for the ball, but I haven't found any way to change these values in the urdf-file.

Thanks in advance.

<details>
My urdf is as follows: 

```
<?xml version="1.0" encoding="utf-8"?>

<robot name="testfloor">
  <mujoco>
    <compiler balanceinertia="true" discardvisual="true" />
    <option cone="elliptic" impratio="10">
      <geom friction="1.5 0.005 0.0001" solimp="0.99 0.99 0.01" solref="0.01 1"/>
    </option>
    
  </mujoco>
  <link name="base_link"/>
  <link name="floor" conaffinity="1" contype="1">
    <visual>
      <geometry>
        <box size="2 2 0.1"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <box size="2 2 0.1"/>
      </geometry>
    </collision>
    
  </link>
  <joint name="floor_j" type="fixed">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="base_link"/>
    <child link="floor"/>
  </joint>

  <link name="ball">
    <inertial>
      <mass value="1.0" />
      <!-- center of mass (com) is defined w.r.t. link local coordinate system -->
      <!-- give a little y-offset to make the top precess -->
      <origin xyz="0 0 0" /> 
      <inertia  ixx="1.0" ixy="0.0"  ixz="0.0"  iyy="1.0"  iyz="0.0"  izz="1.0" />
    </inertial>
    <visual>
      <!-- visual origin is defined w.r.t. link local coordinate system -->
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <sphere radius="0.05"/>
      </geometry>
    </visual>
    <collision>
      <!-- collision origin is defined w.r.t. link local coordinate system -->
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <sphere radius="0.05"/>
      </geometry>
    </collision>
  </link>

  <joint name="ball_j" type="floating">
    <origin rpy="0 0 0" xyz="0 0 1"/>
    <parent link="base_link"/>
    <child link="ball"/>
  </joint>

</robot>
```
</details>

yuvaltassa, 2024-04-10:
Try setting fusestatic to false
https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-fusestatic

santerioksanen, 2024-04-10:
Thanks @yuvaltassa for the suggestion! I should have mentioned earlier, that I already tried this, leading to a different error: `Error: free joint can only be used on top level
Object name = ball, id = 3`

I am unsure what the top-level link is called in this case, tried with `world`, but MuJoCo couldn't find that.

Right now a working intermediate solution seems to be to save the model after loading it in MuJoCo, which somehow gets rid of the issue.

yuvaltassa, 2024-04-11:
I'm glad you have a workaround but what you're saying is disturbing.
You're saying that if you load a URDF the model is not the same as if you save it as MJCF and load it again? Is that correct?
If yes, that's not good. 

What is a MRE for this phenomenon? The model you posted above? Or a different model?

commit d6553ed1 (2024-04-25): Update contype and conaffinity if fusestatic is true. Fixes #1577.

PiperOrigin-RevId: 628073529
Change-Id: I2fe88736d12ec16d72bbd4dce5d349b930147b50
closed 2024-04-25 by copybara-service[bot]