google-deepmind/mujoco #1588: Jacobian computation

FrancescoTrotti, 2024-04-11:
Hi,
I'm trying to compute the forward and inverse differential kinematics of Halfcheetah. 
I'd like to calculate the velocity of the joints given the velocity of the "torso". I'm currently testing to see how to compute the Jacobian.
What I'd expect is that given a velocity of the Halfcheetah joints, obtained by reading the variable data.q_vel after calling mj.step(), I can obtain the relative velocities of the "torso", following this equation:

$$\dot{x} = J \cdot \dot{q}$$

Furthermore, by doing the inverse of the Jacobian and multiplying it by the torso velocities obtained in the previous step, I return exactly to the joint velocities, following this relationship:

$$\dot{q} = J^{-1} \cdot \dot{x}$$

In particular, I am using the function mj_jac(...) (https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-jac)
Below is an example of the code I am using:

```Python
    import mujoco
    import numpy as np

    m = mujoco.MjModel.from_xml_path("half_cheetah.xml")
    d = mujoco.MjData(m)
    d.ctrl = [0.33747295, 0.3996362, 0.18299277, 0.5981481, 0.15231994, 0.07616442]
    mujoco.mj_step(m, d)
    jacp = np.zeros((3, 9))
    jacr = np.zeros((3, 9))
    mujoco.mj_jac(m, d, jacp, jacr, d.xpos[1], 1)
    pe = np.matmul(jacp, np.array(d.qvel))
    pr = np.matmul(jacr, np.array(d.qvel))
    J = np.concatenate((jacp, jacr), axis=0)
    p = np.concatenate((pe, pr), axis=0)
    J_inv = np.linalg.pinv(J)
    q_dot = np.matmul(J_inv, p)

```

With the previous code, I can only obtain the velocities of the first three values of the q_vel vector. 
By modifying or cycling on the variable "body" of mj_jac, however, I do not get the desired velocity vector. How can I solve this?

Thank you

yuvaltassa, 2024-04-11:
Your motivation, as stated, does not make sense.

> I'd like to calculate the velocity of the joints given the velocity of the "torso".

This is clearly impossible, the velocity of the torso does not uniquely determine the velocity of all the joints. Try clarifying what you want and we might be able to help.

FrancescoTrotti, 2024-04-12:
Thank you for your answer.
Yes, you are right I was wrong to indicate the joint name, I meant "root" and not "torso". As far as I know, when we talk about redundant systems, the assignment is not unique, and a given velocity in the Cartesian corresponds to an infinite number of joint configurations. However, if we consider, for instance, a manipulator where we have the Cartesian velocity of the end-effector, it is possible, using the inverse Jacobian, to also trace the velocities relative to the joints. I'm therefore wondering if it is possible in the halfcheetah contest to do something similar: given a velocity at 'root', calculate the relative velocities at the joints. It's currently unclear to me how to calculate the Jacobian correctly. In tests run with the code posted above, I actually run the inverse of the Jacobian multiplied by a velocity in the Cartesian at the "root," and I correctly get back the first three components, which in the documentation should correspond to X, Z sliding velocities and rate of rotation on Y of the "root", while the other velocities at the joints are zero. How can I correctly calculate the complete Jacobian to find the velocities at all joints? In mujoco's documentation, the function mj_jac requires specifying a "const mjtNum point[3]" and an "int body", but I don't understand how to compute the complete Jacobian for all joints.

Thanks again for your help

Balint-H, 2024-04-12:
A very good example of using the jacobian functionality can be found in the LQR tutorial: https://colab.research.google.com/github/deepmind/mujoco/blob/main/python/LQR.ipynb

I recommend browsing through the [API documentation](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#support), as there are a couple variants you may be interested in (e.g. pure mj_jac vs the CoM variants).

The point[3] specified is the global position of a point attached to body with the given index for which the jacobian is evaluated (as different parts of a body will move at different velocities if it is rotating, which is why you need to provide both).

yuvaltassa, 2024-05-05:
> if we consider, for instance, a manipulator where we have the Cartesian velocity of the end-effector, it is possible, using the inverse Jacobian, to also trace the velocities relative to the joints.

Yes, because the velocity of the end effector is a function of the joint velocities, if there is a smooth function, it has a partial derivatives, the Jacobian exists.

>  I'm therefore wondering if it is possible in the halfcheetah contest to do something similar: given a velocity at 'root', calculate the relative velocities at the joints. 

No, it is not possible. The root velocity is not a function of the joint velocities, but rather the other way around. The equivalent to the arm scenario is the Jacobian of the halfchettah's foot w.r.t all joints, including the root. I understand that this would be nice to have, but it is not possible, as stated.

The only thing that would conceivably work is to use a data-based approach: measure lots of data of the moving cheetah and try to fit a non-parametric function to the data that attempts to predict what you want. This may or may not work well, depending on your dataset and ML proficiency. YMMV, good luck.

closed 2024-05-05 by yuvaltassa

Balint-H, 2024-05-05:
@yuvaltassa What if we assume a no-slip condition with the foot/feet in contact with the ground? Would then the end effector Jacobians be closely related to the CoM motion Jacobian (-1 times)?

yuvaltassa, 2024-05-05:
@Balint-H, yes and no, the clue is in your own question. One could assume a no-slip condition with **a foot**, in which case you have a simple kinematic chain, but you **have to pick a foot**. If there are $n>1$ feet touching the ground you have a closed loop which is not analytic. Also, what if you are in a flight phase? Good luck picking a foot in that case 🙂

starimpact, 2025-01-16:
you can use mjd_transitionFD to get sensor Jacobin matrix on relative position.