google-deepmind/mujoco #1589: MuJoCo 3.1.4 modifies the `MjModel.nbvh` value which is not reported in changelogs

pseudo-rnd-thoughts, 2024-04-11:
Thanks for maintaining MuJoCo.

The Gymnasium CI found that one of our environments (Pusher-v4 and Pusher-v5) `nbvh` has changed due to using the latest version of MuJoCo 3.1.4
https://github.com/Farama-Foundation/Gymnasium/actions/runs/8644901511/job/23700995661

Looking through the changelogs for 3.1.4 I don't see any mention of `nbvh` being modified, my guess is that this isn't a bug but I wished to report it in case it was and if possible an explanation of what the change is about in case this affects the environment's behaviour

yuvaltassa, 2024-04-11:
Thanks for noticing!

This is almost certainly due to the optimization in c5aa22b177676928e5f1d7ffdcc2195657fe472c

Since it had no user facing consequences other than a speedup, it did not make it to the changelog.

closed 2024-04-11 by yuvaltassa

pseudo-rnd-thoughts, 2024-04-11:
Thanks @yuvaltassa for the explanation