google-deepmind/mujoco #1590: MJX Python Version Compatibility

willthibault, 2024-04-11:
Hi,

I'm wondering what Python versions are compatible with MJX. I understand that MuJoCo supports Python 3.8+, but I encounter issues using a version lower than 3.10.

When using Python 3.8 and loading the mjx package from a Python shell I get the following error:
```
Python 3.8.10 (default, Nov 14 2022, 12:59:47) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mujoco import mjx
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (2.2.1) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/__init__.py", line 18, in <module>
    from mujoco.mjx._src.collision_driver import collision
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/collision_driver.py", line 22, in <module>
    from mujoco.mjx._src import collision_base
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/collision_base.py", line 22, in <module>
    from mujoco.mjx._src.dataclasses import PyTreeNode
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/dataclasses.py", line 107, in <module>
    class PyTreeNode:
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/dataclasses.py", line 125, in PyTreeNode
    def fields(cls) -> tuple[dataclasses.Field[Any], ...]:
TypeError: 'type' object is not subscriptable
```

This error is no longer an issue in Python 3.9, but another error is encountered:
```
Python 3.9.5 (default, Nov 23 2021, 15:27:38) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mujoco import mjx
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/mujoco/mjx/__init__.py", line 52, in <module>
    from mujoco.mjx._src.test_util import benchmark
  File "/usr/local/lib/python3.9/dist-packages/mujoco/mjx/_src/test_util.py", line 204, in <module>
    joint: str | None = None,
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
```

Is MJX only intended to be used with Python 3.10+ or should this syntax be changed to enable Python 3.8 and 3.9 support?

erikfrey, 2024-04-22:
Ah this is indeed a bug.  We will post a fix shortly.  Thanks for opening the issue.

closed 2024-04-22 by copybara-service[bot]

willthibault, 2024-06-12:
Hi @erikfrey,

I think v3.1.6 introduced another compatibility issue for Python 3.8:

```
Python 3.8.10 (default, Nov 14 2022, 12:59:47) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from mujoco import mjx
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (2.2.1) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/__init__.py", line 18, in <module>
    from mujoco.mjx._src.collision_driver import collision
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/collision_driver.py", line 48, in <module>
    from mujoco.mjx._src.collision_convex import box_box
  File "/usr/local/lib/python3.8/dist-packages/mujoco/mjx/_src/collision_convex.py", line 1053, in <module>
    collider_fn: Callable[[_GeomInfo, _GeomInfo], Collision],
TypeError: 'ABCMeta' object is not subscriptable
```

willthibault, 2024-06-20:
Hi @erikfrey,

I believe adding the import `from typing import Callable` to collision_convex.py fixes the issue. However, `from collections.abc import Callable` is already imported. Their functionality is slightly different, so I wanted to confirm this fix. Otherwise, I'm happy to create a PR for this.

Thanks!

commit c6ab354b (2024-04-22): Remove type annotations unavailable in python < 3.10.  Fixes #1590

PiperOrigin-RevId: 627188913
Change-Id: I9adc627349a4001291ac6d05080b761fee155ab0