google-deepmind/mujoco #1591: Issues with Environment Reset by Loading New XML in MuJoCo for Reinforcement Learning

kankanzheli, 2024-04-12:
Hello MuJoCo Team,

I am currently engaged in reinforcement learning for planning harvest paths, aiming to simulate the complex and variable harvesting environment within a farm. To achieve this, I am considering reloading a new XML file upon each reset during the training process to reflect different environmental setups. I would like to understand if this approach is considered feasible within MuJoCo.

However, I have encountered an issue where the visualization window does not update during the simulation after performing such resets. Below is the implementation for my environment's reset functionality:

`

def merge_xmls(self, main_xml, *additional_xmls):
    main_tree = ET.parse(main_xml)
    main_root = main_tree.getroot()
    body_names = []
    main_tendon = main_root.find("tendon")
    if main_tendon is None:
        main_tendon = ET.SubElement(main_root, "tendon")
    for add_xml in additional_xmls:
        add_tree = ET.parse(add_xml)
        add_root = add_tree.getroot()
        for elem in add_root.find("worldbody"):
            main_root.find("worldbody").append(elem)
            body_names.append(elem.get('name')) 
        add_tendon = add_root.find("tendon")
        if add_tendon is not None:
            for spatial in add_tendon:
                main_tendon.append(spatial)
    self.place_random_obstacle_and_target(main_root)
    merged_path = "merged.xml"
    with open(merged_path, "wb") as f:
        f.write(ET.tostring(main_root))
    return merged_path, body_names

def reset(self, seed=None, return_info=False, options=None):
    if seed is not None:
        np.random.seed(seed)
    self.merged_xml, self.body_names = self.merge_xmls(self.robot_xml, self.ball_xml, self.box_xml)
    self.model = mujoco.MjModel.from_xml_path(self.merged_xml)
    self.data = mujoco.MjData(self.model)
    mujoco.mj_forward(self.model, self.data)
    ball_id = mujoco.mj_name2id(self.model, mujoco.mjtObj.mjOBJ_BODY, "target")
    ball_xpos = self.data.xpos[ball_id]
    #self.reset_robot_state()
    if self.viewer is not None:
        self.viewer.sync()
    self.step_count = 0
    observation = self.get_observation()   
    info = {}  
    return observation, info `

And here is the code for the visualization part:

`def render(self, mode="none"):
    if mode == "human":
        if self.viewer is None:
            self.viewer = mujoco.viewer.launch_passive(self.model, self.data)

if name == "main":
    env.reset()
    done = False
    env.render(mode="human") 
    while not done:
        action = env.action_space.sample()
        obs, reward, done, truncated, info = env.step(action)
        if done or truncated:
            obs = env.reset()
            done = False`

In the step method, I ensure that the simulation state is updated, and I call mj_step and self.viewer.sync() at the end of each step. 
The issue I am facing is that objects within the visualization window remain static and do not reflect the updated state after the reset. Could this be due to the reloading of the XML file, and if so, is there a recommended approach for handling such dynamic environment changes in MuJoCo?

Any guidance or recommendations would be greatly appreciated.

Best regards

kevinzakka, 2024-04-15:
Yuval is correct. One simple workaround is to call `self.viewer._sim().load(model, data, "")` after you recompile the model and data.

kankannali, 2024-04-17:
Thank you for your help. I solved my problem by executing self.viewer._sim().load(model, data, "") every time when I reset the environment.

closed 2024-04-17 by kevinzakka