google-deepmind/mujoco #1593: Instability in simulation: `Got MuJoCo Warning: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. `

MasterXiong, 2024-04-14:
Hi,

I'm a student and I'm trying to use MuJoCo version 2.1 for RL training on locomotion tasks.

I'm looking for some help with an unstable issue during simulation. When I run the RL training or simply run simulation on the robot with random actions, sometimes I got this warning: `Got MuJoCo Warning: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. ` I checked the answers to some similar issues before, but none of them seems to solve my problem here. Could you please help have a look on what may be the reason of the instability here? Thanks a lot!

Below is the XML of an example robot that encountered this issue. 

```XML
<?xml version='1.0' encoding='UTF-8'?>
<!-- Universal Animal Template: unimal -->
<mujoco model="unimal">
  <compiler angle="degree"/>
  <size njmax="2000" nconmax="500"/>
  <option timestep=".005">
    <flag filterparent="disable"/>
  </option>
  <!-- Common defaults to make search space tractable -->
  <default>
    <!-- Define motor defaults -->
    <motor ctrlrange="-1 1" ctrllimited="true"/>
    <!-- Define joint defaults -->
    <default class="normal_joint">
      <joint type="hinge" damping="1" stiffness="1" armature="1" limited="true" range="-120 120" solimplimit="0 0.99 0.01"/>
    </default>
    <default class="walker_joint">
      <joint type="hinge" damping="0.2" stiffness="1" armature=".01" limited="true" range="-120 120" solimplimit="0 0.99 0.01"/>
    </default>
    <default class="stiff_joint">
      <joint type="hinge" damping="5" stiffness="10" armature=".01" limited="true" solimplimit="0 0.99 0.01"/>
    </default>
    <default class="free">
      <joint limited="false" damping="0" armature="0" stiffness="0"/>
    </default>
    <default class="growth_site">
      <site size="1e-6 1e-6 1e-6"/>
    </default>
    <default class="torso_growth_site">
      <site size="1e-6 1e-6 1e-6"/>
    </default>
    <default class="mirror_growth_site">
      <site size="1e-6 1e-6 1e-6"/>
    </default>
    <default class="btm_pos_site">
      <site size="1e-6 1e-6 1e-6"/>
    </default>
    <default class="box_face_site">
      <site size="1e-6 1e-6 1e-6"/>
    </default>
    <default class="imu_vel">
      <site type="box" size="0.05" rgba="1 0 0 1"/>
    </default>
    <default class="touch_site">
      <site group="3" rgba="0 0 1 .3"/>
    </default>
    <default class="food_site">
      <site material="food" size="0.15"/>
    </default>
    <!-- Define geom defaults -->
    <geom type="capsule" condim="3" friction="0.7 0.1 0.1" material="self"/>
  </default>
  <worldbody>
    <light diffuse="1 1 1" directional="true" exponent="1" pos="0 0 1" specular=".1 .1 .1"/>
    <!-- <geom name="floor" type="plane" pos="0 0 0" size="50 50 1" material="grid"/> -->
    <!-- Programatically generated xml goes here -->
    <body name="torso/0" pos="0 0 0.65">
      <joint name="root" type="free" class="free"/>
      <site name="root" class="imu_vel"/>
      <geom name="torso/0" type="sphere" size="0.1" condim="3" density="1000"/>
      <camera name="side" pos="0 -7 2" xyaxes="1 0 0 0 1 2" mode="trackcom"/>
[... 131 lines cut ...]
    <material name="decoration" rgba=".3 .5 .7 1"/>
    <material name="eye" rgba="0 .2 1 1"/>
    <material name="target" rgba=".6 .3 .3 1"/>
    <material name="target_default" rgba=".6 .3 .3 1"/>
    <material name="target_highlight" rgba=".6 .3 .3 .4"/>
    <material name="site" rgba=".5 .5 .5 .3"/>
    <material name="ball" texture="ball"/>
  </asset>
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1="0.1 0.1 0.1" rgb2="0.1 0.1 0.1" width="300" height="300" mark="edge" markrgb="0.2 0.2 0.2"/>
    <texture name="hfield" type="2d" builtin="checker" rgb1="0.1 0.1 0.1" rgb2="0.1 0.1 0.1" width="300" height="300"/>
    <texture name="wall" type="2d" builtin="flat" rgb1="0.9 0.7 0" rgb2="0.9 0.7 0" width="300" height="300"/>
    <texture name="platform" type="2d" builtin="flat" rgb1="0.3 0 0.8" rgb2="0.3 0 0.8" width="300" height="300"/>
    <texture name="boundary" type="2d" builtin="flat" rgb1="0.3 0.3 0.3" rgb2="0.3 0.3 0.3" width="300" height="300"/>
    <texture name="jump" type="2d" builtin="flat" rgb1="0.3 0.3 0.3" rgb2="0.3 0.3 0.3" width="300" height="300"/>
    <texture name="skybox" type="skybox" builtin="flat" rgb1="0.8 1 1" rgb2="0.8 1 1" width="800" height="800"/>
    <texture name="ball" builtin="checker" mark="cross" width="151" height="151" rgb1="0.1 0.1 0.1" rgb2="0.9 0.9 0.9" markrgb="1 1 1"/>
  </asset>
  <visual>
    <headlight ambient=".4 .4 .4" diffuse=".8 .8 .8" specular="0.1 0.1 0.1"/>
    <map znear=".01"/>
    <quality shadowsize="2048"/>
  </visual>
</mujoco>
```

yuvaltassa, 2024-04-15:
You read this?
https://mujoco.readthedocs.io/en/latest/overview.html#divergence

MasterXiong, 2024-04-16:
Hi @yuvaltassa , thanks for your reply! The link is quite helpful. 
Sorry that I'm not quite familiar with Mujoco simulation, so I was wondering that if you could help answer some further questions? As I'm not sure if my own understanding is correct. Thanks!
1. The link suggests to use a smaller timestep, but at the cost of slower simulation. I don't quite get why using a smaller timestep will slow down the simulation? And how will using a smaller timestep affect the RL training process? Does one timestep in an MDP episode correspond to integration over the timestep specified in the XML file in simulation?
2. The link also suggests switching to a more stable integrator. Do you have some suggestions on which integrator will be more stable and most likely to solve my issue here? And will it slow down the simulation a lot?
3. If the divergence still happens after these fixes, is there some common way to detect and drop these divergent simulation data when doing RL training (specifically I'm using PPO)?

Thanks a lot for your help!

quagla, 2024-04-22:
1. The simulation time of a single step is essentially constant, so if you take more steps per unit of physical time, you can simulate less physical time per simulation time (hence slower sim).
2. Implicit methods are more stable.
3. Probably but I'll let @yuvaltassa comment on this.

yuvaltassa, 2024-04-22:
Upon detecting divergence, MuJoCo automatically resets the simulation and writes into `mjData.warning`. This is documented [here](https://mujoco.readthedocs.io/en/latest/computation/index.html#stages). So your options are:
1. Check if `data.warning[mujoco.mjtWarning.mjWARN_BADQACC].number > 0`.
2. Check if `mjData.time` is 0, not at the start of the episode.

closed 2024-04-22 by yuvaltassa

MasterXiong, 2024-04-22:
Thanks for your help!

I already filtered out unstable trajectories by detecting warning signals as suggested. However, even if I filter out these trajectories with warnings, the PPO training is still not stable. I was wondering that is it possible that there are still some rollouts with large values after integration, but they are just not significant enough to trigger a warning? What is the threshold here to determine if a warning will be raised?