google-deepmind/mujoco #1595: Problems on loading XML for V3.1.4

HappyLamb123, 2024-04-15:
I recently updated my MuJoCo to the latest version but encountered some problems. I created a loop to recursively update my XML files, specifically aiming to perform real-time adjustments to the ground by updating the XML. However, with version 3.1.4, it failed to read the XML from the path after around 10 iterations. I tried both relative and absolute paths, but neither worked. When I downgraded my MuJoCo back to version 3.1.3, the issues were resolved. Could you please investigate those problems?

vikashplus, 2024-04-16:
Here are a few other signals I have 

1. There is a bug while saving XMLs with frames 
original model 
``` XML 
<mujoco model="good">

<worldbody>
    <geom type="plane" size=".1 .1 1"/>

    <body name="root" pos="0 0 .5">
       <geom type="sphere" size=".01" rgba="1 0 0 1"/>
       <joint name="slide1" type="slide" limited="true" range="-1 1" actuatorfrclimited="false"/>

      <frame name="fr1" pos="0 0 .01">
        <geom name="geom_fr1" type="sphere" size=".01" rgba="1 1 0 1"/>
      </frame>

      <frame name="fr2" pos="0 0 -.01">
        <geom name="geom_fr2" type="sphere" size=".01" rgba="1 1 0 1"/>
      </frame>

    </body>

</worldbody>
</mujoco>
```
Loading and saving the model using following with break the model due to difference in frame names 
```python 
mjc_model = mujoco.MjModel.from_xml_path(model_path)
mujoco.mj_saveLastXML(model_save_path, mjc_model)
model = mujoco.MjModel.from_xml_path(model_save_path)
``` 

Saved model (bug)
``` XML 
<mujoco model="bug">

<worldbody>
    <geom type="plane" size=".1 .1 1"/>

    <body name="root" pos="0 0 .5">
       <geom type="sphere" size=".01" rgba="1 0 0 1"/>
       <joint name="slide1" type="slide" limited="true" range="-1 1" actuatorfrclimited="false"/>

      <frame name="root" pos="0 0 .01">
        <geom name="geom_fr1" type="sphere" size=".01" rgba="1 1 0 1"/>
      </frame>

      <frame name="root" pos="0 0 -.01">
        <geom name="geom_fr2" type="sphere" size=".01" rgba="1 1 0 1"/>
      </frame>

    </body>

</worldbody>
</mujoco>
```

vikashplus, 2024-04-16:
2. There is a bug while saving binary models 

```python
mjc_model = mujoco.MjModel.from_xml_path(model_path)
mujoco.mj_saveModel(mjc_model, model_path[:-4]+"_saved.mjb")
```

vikashplus, 2024-04-16:
3. For some models, the saved models(right) are different than the original models(left).  
[image]
[image]

yuvaltassa, 2024-04-16:
@vikashplus i suspect these are different bugs. Can you try to make a MRE of the second bug please? Thanks.

yuvaltassa, 2024-04-17:
@vikashplus these are 3 separate bugs (it's not helpful that they are described in the same issue).

1. Will be fixed shortly.
2. Will be fixed shortly. A workaround for now is to add None `mujoco.mj_saveModel(mjc_model, filename, None) `.
3. We need a MRE for this.

yuvaltassa, 2024-04-17:
1. was fixed by 4b6c07cd7d7c80767439b805cc31ff0947ca7677
2. was fixed by 2b4975811b4273415675ad6ce9753adb1637996a
3. We need an MRE.

Closing this issue. @HappyLamb123, @vikashplus: if your issue is not resolved at head (i.e. please build from source and check), please open an new issue with a Minimum Reproducible Example.

closed 2024-04-17 by yuvaltassa