google-deepmind/mujoco #1598: Geom Mesh Transform from local .obj coords to global coords

MasonJohnHawver42, 2024-04-16:
Hi,

I'm a student and I'm trying to use MuJoCo for data generation, first a scene's physics is performed in mujoco and the absolute transforms for each geom visual mesh is stored to be rendered in another program. 

I'm looking for some help with finding the absolute transform from the coordinates stored in the meshes vertex buffer to the global position they are at in the physics simulation. I know about geom_xpos, geom_xmat, mesh_pos, and mesh_quat. I need helping putting this data together to make a 4x4 model matrix that can be used in another application. I also need to find the value for the meshes scale in mujoco, right now its just hardcoded. 

here is the result on a robotic gripper:
[image]

yuvaltassa, 2024-04-16:
Did you actually search the issues?

See #345 

We just added mesh_scale, see latest [changelog](https://mujoco.readthedocs.io/en/latest/changelog.html)

closed 2024-04-16 by yuvaltassa

MasonJohnHawver42, 2024-04-16:
Thank you for reffering me to that issue and changelog. I have seen that issue before and here is what I have attempted:

```python
model_id = robot_model_ids[model_index]

scale = [.001, .001, .001, 1] # coppied from xml, I'll use mesh_scale when its avaliable

scale_matrix = np.diag(scale)

geom_pos = data.geom_xpos[geom_id]
geom_rot = data.geom_xmat[geom_id].reshape(3, 3)

model_pos = model.mesh_pos[model_id]
model_quat = model.mesh_quat[model_id]
model_rot = R.from_quat(model_quat).as_matrix()

rot_matrix_geom = np.eye(4)
rot_matrix_geom[:3, :3] = geom_rot

translation_matrix_geom = np.eye(4)
translation_matrix_geom[:3, 3] = geom_pos

rot_matrix_model = np.eye(4)
rot_matrix_model[:3, :3] = model_rot

translation_matrix_model = np.eye(4)
translation_matrix_model[:3, 3] = model_pos

pose = translation_matrix_geom @ rot_matrix_geom @ translation_matrix_model @ rot_matrix_model @ scale_matrix
robot_poses[model_index] = pose
```

The pose is then applied directly to the verticies read in from a .stl file, and produced the result above. 

I have also tried:

```python
pose = translation_matrix_geom @ rot_matrix_geom @ scale_matrix
pose = translation_matrix_geom @ rot_matrix_geom @ scale_matrix @ translation_matrix_model @ rot_matrix_model
...
```

What information am I missing, or am I composing these transforms incorrectly. Also quats are represented as (w, x, y, z) right?

Thank you for your paitence adn quick response.

MasonJohnHawver42, 2024-04-16:
Also could anybody point me towards where in your code for visualization you render meshes, I can't find it.

yuvaltassa, 2024-04-16:
> The pose is then applied directly to the verticies read in from a .stl file

You can't do that without first applying the transforms that MuJoCo applied. This can only work for the vertices in mjModel, which are post-transform. Read [this](https://mujoco.readthedocs.io/en/latest/XMLreference.html#asset-mesh). The actual transform, is you want to recapitulate it, is in `mjModel.mesh_{scale,pos,quat}`, but you can just take the vertices from mjModel. Also, I don't know what this R thing is, please use MuJoCo's [built in functions](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mju-mat2quat) to avoid different conventions.

> Also quats are represented as (w, x, y, z) right?

Yes.