google-deepmind/mujoco #1599: [question]

hg6185, 2024-04-16:
Hi,

I'm a student and I'm trying to use MuJoCo for a personal project. I am using the panda.xml from mujoco menagerie. I added a site to reference the end effector position of the robot. However I have difficulties to interpret the values of site_xpos and site_xmat.

From the docs, I read that site_xpos.shape = (1,3) and site_xmat.shape=(1,9), given that you have only one site.
How do I compute the orientation from xmat? I read that these are the sine cosine values? But why is than the position not the same with xpos?
To reformulate my question: What is x_mat and how can I compute the real orientation?

Bonus question: The angles from the computation of jacr using mj_jac are in radiants?

v-r-a, 2024-04-16:
Hi, I am a user. I hope the following helps:

0) I see that there is no site by default in the panda model. If you have added one (say to `hand` body) yourself,

1) The matrix `data.site_xmat`  (defined in row-major format) defines the orientation of the site frame. The columns of the rotation matrix are the [direction cosines](https://en.wikipedia.org/wiki/Direction_cosine) of the site axes. From the rotation matrix, you can describe the orientation using Euler parameters/quaternion, Euler angles or the axis-angle notation. [This section](https://en.wikipedia.org/wiki/Rotation_matrix#In_three_dimensions) on Wikipedia may help you. MuJoCo has a `mat2quat` [function](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mju-mat2quat) that may help you. You will have to write other conversions yourself.

2) Not sure about your `site_xpos` question. It would give you the position of the site in the world frame.

3) The absolute angular velocity of the site in the world frame is given by: $\omega$ = [jacr] * `data.qvel`. You can see that elements in `jacr`  would be dimensionless if all the degrees-of-freedom are revolute joints for example. You can do the dimension analysis to find the units of elements in `jacr` if there are prismatic joints in the model.

hg6185, 2024-04-17:
Thanks so much!!! That helped a lot! I was not thinking in robotics. I only had the angles in mind :)
For completeness, the [link](https://en.wikipedia.org/wiki/Rotation_matrix#General_3D_rotations) to the general 3D  rotation matrix.

I have a slightly related question for you, as you seem to be very knowledgeable of Mujoco:
I build a scene with 4 additional objects. One is a free joint. Is there any nice way, to grab only my panda?
For instance, my jacp has the dimension of (3,15), which for understanding purposes and computing the inverse kinematics is a bit annoying. Is there a best practice?

v-r-a, 2024-04-17:
If you haven't found it already, see discussion #1419. 

If you are writing in Python, https://github.com/kevinzakka/mjctrl would be a good starting point for your code.

closed 2024-04-17 by hg6185

hg6185, 2024-04-17:
Thanks again. mjctrl is exactly the point to start that I needed!