google-deepmind/mujoco #1600: A resistiv torque on joint

Alesel-9, 2024-04-17:
Hello im asking this question since i didn't understand the answer given in [issue#1095](https://github.com/google-deepmind/mujoco/issues/1095)

Similarly to that issue im trying to find a way to see how much torque is exerted on a hinge type joint. 
The reason im trying to see this is to use a general type actuator to produce a resistiv torque as to simulate a damping moment.
For clarity see this model:
```

 <worldbody>
    <geom type="plane" size="0 0 .01" material="grid" friction="1 1 0.1"/>
    <!--Pendel Arm-->
    <body name="main" pos="0 0 4" euler="-0 0 0"  >
      <freejoint/>
      <geom class="body" rgba=" 0.2 0.3 0 1" />
      <inertial mass="10900" pos="0.3 0 0.8" diaginertia="0.000734064 0.000468438 0.000398719"/>
      <light name="top light" pos="0 0 2" mode="trackcom" diffuse=".4 .4 .4"/>

      <body name="parm1" pos="-2.8 -1.2 0.5" euler="0 45 0 ">
          <site name="fls" pos="0.4 0 0"/>
          <joint name="jparm1" type="hinge"  axis="0 1 0" limited="true" range="-45 45"  stiffness="17000"  springref="15" />
          <geom name = "parm1" type="capsule" fromto="0 0 0 0.7 0 0" size="0.07" rgba="0.5 0.2 0 1"/>
          <!--Wheel -->
          <body pos="0.7 -0.1 0" euler="90 0 0"  >
              <joint name="jw1" type="hinge" axis="0 0 1" frictionloss="10" actuatorfrcrange="-0.5 0.5"/>
              <geom name = "w1" type="cylinder"  fromto="0 0 0 0 0 0.25" size="0.35" rgba=".05 .2 .1 1" friction="1 1 0.1"/>
          </body>
      </body>
  </worldbody>

  <actuator>
      <general name="damper" joint="jparm1" ctrlrange="0 15000" />
      <velocity name= "move" joint="jw1" ctrlrange="0 50"/>
  </actuator>`
```
My goal is to see the torque that is exerted upon the joint and then apply a counter torque from the general acuator.

My question is to understand what the **#data.joint("parm1").qfrc_constraint** and **data.joint("parm1").qfrc_smooth** actually gives me. I've looked through the documentation but it is only mention sparsly what it actually shows, so here I am.
Im working in python, and the code above is not to be run, it's only for referens. Please ask me to clarify if i didn't explain properly

yuvaltassa, 2024-04-17:
> The reason im trying to see this is to use a general type actuator to produce a resistiv torque as to simulate a damping moment.
 
 You know about [dampers](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-damper), yes?
 
 `qfrc_{smooth, constraint}` are the smooth forces and constraint forces, respectively. You seem to also want the forces applied by the other actuator, so look at `qfrc_actuator` or `actuator_force`. It might help to read the file [engine_forward.c](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_forward.c), this is the high level pipeline.

closed 2024-04-17 by yuvaltassa

Alesel-9, 2024-04-18:
@yuvaltassa 
Well yes, I do know about dampers but they are prportional to the velocity of the joint and that is not what I seek to do. 
I looked at [engine_forward.c](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_forward.c) but it didn't tell me what it calculates, it actually made be more confused since at [line 951](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_forward.c#L469) there is a summation of the the two value, which as seen in my 'Edit' section above would be equal to 0. 

What do you mean when you say 'smooth forces and constraint forces'?
And how do they relate to the torque of the joint?

Alesel-9, 2024-04-18:
After testing around a bit more i have made the following observations:

 **qfrc_smooth** is the total torque that the joint imposes on itself. For example, in the model above there is a stiffness property defined in the joint, the tourque applied by that spring on the joint is what shows up in  **qfrc_smooth** .  Additionally if there are other torque inducing properties applied on the joint they would too be summed up into   **qfrc_smooth** and displayed as an equivalent  torque.

 **qfrc_constraint** is essentially the opposite. It displays the external torque applied to the joint. For example; in the model above there is a moment arm extanding from the joint and then makes contact with the ground. The normal force that the ground exerts at the end of the moment arm is converted to torque around the joint, but in the opposite direction to  the torque that the spring induces. 

That is why, at equlibrium the two values  **qfrc_smooth** and **qfrc_constraint** would be equal, allbeit opposite sign. Becuse (in the context of the above model) the sum of the internal torque (the spring pressing the moment arm into the ground) and the external torque (the normal force on the moment arm) would be 0 since they have reached a steady state and pushing the joint to rotate in different directions.

Im I understanding this correctly?
If not, where am I wrong?        
@yuvaltassa

yuvaltassa, 2024-04-18:
More or less, yes.

[Here](https://github.com/google-deepmind/mujoco/blob/4b7b40d3295bd1935b93af3d38eaaf3c1fc4ec1e/src/engine/engine_forward.c#L521) is the definition of **qfrc_smooth**: the sum of **bias** (Coriolis, centripetal, gravitational) forces, **passive** (spring, damper, fluid) forces and **actuation** forces (both modeled with native actuators and directly applied).

**qfrc_constraint** is the projection of **constraint forces** onto joint space. These include **contacts**, **equalities** and **joint limits**. [Here](https://github.com/google-deepmind/mujoco/blob/4b7b40d3295bd1935b93af3d38eaaf3c1fc4ec1e/src/engine/engine_forward.c#L813) is where these two components are added together to get the total force $f$ and the equation $Ma=f$ is solved for the acceleration $a$.

yuvaltassa, 2024-04-19:
Yes. That's what `qfrc_applied` is for: applied actuation forces that are outside of the standard actuation framework.

Alesel-9, 2024-04-20:
@yuvaltassa 
Can you provide an example for **qfrc_applied** or explain in a different way. 
I understand that there is a contrediction by using the sum of **qfrc_smooth** and **qfrc_constraint** to set a resistiv torque, and therefore i need to use **qfrc_applied** . But I don't understand what  **qfrc_applied** gives me or what it represents.

When i tried the following'.

```
import mujoco as mj
model = mj.MjModel.from_xml_path('MAIN_Scene_v2.xml')
data=mj.MjData(model)
mj.mj_resetData(model,data)
print(data.joint('parm1'))
data.actuator('damper').ctrl=3000

for i in range(1000):
    mj.mj_step(model, data)
print(data.joint('parm1'))  
```

I got this:

```

<_MjDataJointViews
  cdof: array([[0., 0., 0., 0., 0., 0.]])
  cdof_dot: array([[0., 0., 0., 0., 0., 0.]])
  id: 1
  name: 'jnt_pendel_arm_1'
  qLDiagInv: array([0.])
  qLDiagSqrtInv: array([0.])
  qacc: array([0.])
  qacc_smooth: array([0.])
  qacc_warmstart: array([0.])
  qfrc_actuator: array([0.])
  qfrc_applied: array([0.])
  qfrc_bias: array([0.])
  qfrc_constraint: array([0.])
  qfrc_inverse: array([0.])
  qfrc_passive: array([0.])
  qfrc_smooth: array([0.])
  qpos: array([0.])
  qvel: array([0.])
  xanchor: array([0., 0., 0.])
  xaxis: array([0., 0., 0.])
>
<_MjDataJointViews
  cdof: array([[0., 0., 0., 0., 1., 0.]])
  cdof_dot: array([[0., 0., 0., 0., 0., 0.]])
  id: 1
  name: 'jnt_pendel_arm_1'
  qLDiagInv: array([0.02269422])
  qLDiagSqrtInv: array([0.15064602])
  qacc: array([-0.95754868])
  qacc_smooth: array([232.76126695])
  qacc_warmstart: array([-0.95754868])
  qfrc_actuator: array([2999.13293061])
  qfrc_applied: array([0.])
  qfrc_bias: array([-495.68075765])
  qfrc_constraint: array([-12582.45606242])
  qfrc_inverse: array([0.])
  qfrc_passive: array([9099.06195486])
  qfrc_smooth: array([12593.87564312])
  qpos: array([-0.10043278])
  qvel: array([0.86515429])
  xanchor: array([ 1.21370952, -3.00947769,  1.0609724 ])
  xaxis: array([-9.99997867e-01, -1.87180061e-03,  8.73684102e-04])
>

```

The issue is that **qfrc_applied** is zero in both cases so I don't understand what it shows and how it's supposed to be used?

yuvaltassa, 2024-04-21:
`qfrc_applied` if for the user (you). Just write into it whatever applied force you want.

Alesel-9, 2024-04-21:
@yuvaltassa 
So just for confirmention; I use **qfrc_smooth** and **qfrc_constraint** to measure the total torque in the joint and then use **qfrc_applied** to set the resulting resistiv torque? Like this, if the total torque in the joint is +3014:
`data.joint.qfrc_applied('parm1')=-3014`
and this will apply a 3014 Nm torque in the opposite direction to the joint 'parm1', free of contradictions and such?

yuvaltassa, 2024-04-21:
Yes