google-deepmind/mujoco #1606: Exceeding Memory Limits: Cannot Simulate a Soft-Body/flexcomp object with 36624 voxels

kurtenkera, 2024-04-19:
Hi, 

I have attached an .msh file of an object that I'm trying to simulate as a soft-body/`flexcomp` in MuJoCo (
[3dmbb.zip](https://github.com/google-deepmind/mujoco/files/15033710/3dmbb.zip)). When I try to execute the code below:

```
import mujoco
import mediapy as media

xml = r"""
<mujoco model="3DMBB">
  <worldbody>
    <flexcomp name="3DMBB" type="gmsh" dim="3" file="3dmbb.msh">
    <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
"""

I get the following error message:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[1], [line 20](vscode-notebook-cell:?execution_count=1&line=20)
      [9](vscode-notebook-cell:?execution_count=1&line=9) xml = r"""
     [10](vscode-notebook-cell:?execution_count=1&line=10) <mujoco model="3DMBB">
     [11](vscode-notebook-cell:?execution_count=1&line=11)   <worldbody>
   (...)
     [16](vscode-notebook-cell:?execution_count=1&line=16) </mujoco>
     [17](vscode-notebook-cell:?execution_count=1&line=17) """
     [19](vscode-notebook-cell:?execution_count=1&line=19) # Make model and data
---> [20](vscode-notebook-cell:?execution_count=1&line=20) model = mujoco.MjModel.from_xml_string(xml)
     [21](vscode-notebook-cell:?execution_count=1&line=21) data = mujoco.MjData(model)
     [23](vscode-notebook-cell:?execution_count=1&line=23) # Make renderer, render and show the pixels

ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 7815721488, available = 7812204224, requested = 8100608328 (ne = 56110, nf = 0, nefc = 56110, ncon = 0)
```

Is the only way around this to re-mesh my object with fewer elements (i.e., make the element size bigger)? If so, how will having a less fine-grained mesh affect the simulation accuracy?

yuvaltassa, 2024-04-19:
Did you try this?
https://mujoco.readthedocs.io/en/latest/XMLreference.html#size-memory
https://mujoco.readthedocs.io/en/latest/modeling.html#csize

kurtenkera, 2024-04-21:
@yuvaltassa The mesh file that I'm currently trying to load is attached here (
[MSH41_exceed_memory.zip](https://github.com/google-deepmind/mujoco/files/15051056/MSH41_exceed_memory.zip)). I tried the advice that you recommended in those above links. As you can see in the code below, I include the line `<size memory = "8G"/>` in my XML file which I thought would be a reasonable amount of allocated memory:

```
import mujoco
import mediapy as media

xml = r"""
<mujoco model="3DMBB">
  <size memory = "8G"/>
  <worldbody>
    <flexcomp name="3DMBB" type="gmsh" dim="3" file="converted_3d_mbb_ascii.stl_out_vol.msh">
    <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
"""

However, I get this error message below:

```
ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 8573357512, available = 107779936, requested = 8461964232 (ne = 57810, nf = 0, nefc = 57810, ncon = 0)
```

How do I load in .msh files with a large amount of nodes and elements (~10k nodes, ~40k elements) like this one? Do I need to try run the model on a different machine with more RAM (my machine has 64GB of memory)? Even when I try to load the model with the following setting `<size memory = "20G"/>` to increase the memory allocation I get the following output:

```
ValueError: XML Error: unsigned integer with an optional suffix {K,M,G,T,P,E} is expected in attribute 'memory' (or the size specified is too big)
Element 'size', line 3
```

This indicates that I can't allocate 20GB of memory even though I should be able to on my machine?

yuvaltassa, 2024-04-21:
Ah the latter seems like an actual bug in the parser, we should be using `long` instead of `int` somewhere...

kurtenkera, 2024-04-22:
@yuvaltassa Thanks for the clarification - what about the first/former issue? Why do I get the error below when it clearly says I have both a `max` and `available` amount of memory greater than what was requested?

```
ValueError: Error: engine error: mj_stackAlloc: insufficient memory: max = 8573357512, available = 107779936, requested = 8461964232 (ne = 57810, nf = 0, nefc = 57810, ncon = 0)
```

 Will I have to wait for the bug that you identified to be fixed before I can model this 3D mesh file in MuJoCo?

yuvaltassa, 2024-04-22:
> Why do I get the error below when it clearly says I have both a max and available amount of memory greater than what was requested?

Count more carefully, `available` is smaller by an order of magnitude...

kurtenkera, 2024-04-22:
So when I requested 8GB of memory in the above example, I actually only had 100MB available? Why is this the case? When I print out the available memory on my machine in the terminal, it says I currently have 39GB of memory available (see below):

[image]

Surely 39GB of available memory should be more than enough to simulate the mesh file with ~10k nodes, ~40k elements, no?

yuvaltassa, 2024-04-22:
Like I said, I think we have some bug related to using an `int` somewhere where we shouldn't. `INT_MAX` is 2147483647 which is just 2.14GB ...

closed 2024-04-22 by copybara-service[bot]

kurtenkera, 2024-04-23:
When will the next release be?

yuvaltassa, 2024-04-23:
May 6th

commit 546a27ca (2024-04-22): Fix memory allocation error when memory size is larger than 2G. Fixes #1606

PiperOrigin-RevId: 627058355
Change-Id: Iebef647a0107d1f112a876d705f5b3ea888b4a41