google-deepmind/mujoco #1607: call mj_setConst on a mjx model after changing body masses, inertias for domain randomization

JeyRunner, 2024-04-19:
Hi, I am a student using mjx in combination with brax for RL.
For implementing domain randomization I followed the example notebook which look like this:
```python
def domain_randomize(sys, rng):
  """Randomizes the mjx.Model."""
  @jax.vmap
  def rand(rng):
    # randomize frictions ...
    return friction, gain, bias
  friction, gain, bias = rand(rng)

  in_axes = jax.tree_util.tree_map(lambda x: None, sys)
  in_axes = in_axes.tree_replace({
      'geom_friction': 0,
  })

  sys = sys.tree_replace({
      'geom_friction': friction,
  })

  return sys, in_axes
```

Now I also want to randomize over body masses and inertias, but according to https://github.com/google-deepmind/mujoco/issues/764#issuecomment-1464708759 this requires adapting dependent constants like `dof_M0` ...

With the normal mujoco (non mjx) pipeline the solution seems to be calling `mj_setConst` after changing the link masses/inertias to have the dependent constant recalculated.

With mjx this is not possible since, to my understanding, there is no way of copying an mjx model back to a mujoco model (like an `mjx.get_model` function).
When this function would exist, the workflow could look like this:
 - randomizing the batched mjx model with the domain_randomize function using jax.vmap
 - copying the batched model to an array of mujoco models
 - call  `mj_setConst` on each mujoco model
 - copy back the fixed list of mujoco models to a batched mjx model

Or is there another way to call `mj_setConst` directly on a batched mjx model?

lkoelman, 2024-05-08:
Bumping into the same problem. Having a solution to this issue would definitely make the domain randomisation functionality much more useful.

JeyRunner, 2024-05-09:
1. I guess the best solution would be to port everything in https://github.com/google-deepmind/mujoco/blob/1d181786a25da07a9ac536ae0e78291b221e27f0/src/engine/engine_setconst.c#L506C6-L506C17 to mjx/jax.
But I am not sure how much work this would be.

2. In the meantime, the copy model method could be used as a workaround. However, this may have significant performance disadvantages. Especially when used in an RL training loop where domain randomization is performed each reset call over many environments.

3. Another workaround may be to call `mj_setConst(mjModel* m, mjData* d)` via [ `jax.experimental.io_callback`](https://jax.readthedocs.io/en/latest/notebooks/external_callbacks.html#flavors-of-callback) from within the jitted `domain_randomize` function, but this again requires some functionality to copy an mjx model to a normal mujoco model (`mjx.get_model`). At least this would provide the same simple API as (1.) (which can be called from within vmap) but without the need to fully port `mj_setConst(mjModel* m, mjData* d)`.

btaba, 2024-05-22:
Thanks for opening the bug @JeyRunner! Option 1 sounds the best to me. Feel free to open up a PR! Indeed this functionality would be really useful for randomizing mass properties (in a correct way).

hshi74, 2024-09-04:
Having the same question. Are there any updates on this?

btaba, 2024-09-09:
@hshi74 Apologies there hasn't been movement on this issue yet.

As a workaround, I recommend creating the randomized models in MuJoCo so that fields get compiled correctly, and doing a `put_model` on each randomized model. You can then tree_map/concat the models in MJX. During training, avoid calling resets on the `mjx.Model` (for fields that need to be re-compiled). We find that many policies do fine with a large batch of reset states that get set once at the beginning of training. Let us know if there are any updates otherwise!

amacati, 2024-12-29:
I'm also running into this problem and would be happy about any progress on this.

> @hshi74 Apologies there hasn't been movement on this issue yet.
>
> As a workaround, I recommend creating the randomized models in MuJoCo so that fields get compiled correctly, and doing a `put_model` [...]

One issue with this approach is that batching full models wastes resources on properties that remain untouched. The neat thing with the randomization described in the example notebook is that only `mjx.Model.geom_friction` receives an additional batch dimension, while `mjx.Model.body_mass` and other fields of `Model` are not copied. 

```python
import jax
import mujoco
from mujoco import mjx

xml = """
<mujoco>
  <worldbody>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""
mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)
mjx_model = mjx.put_model(mj_model)
mjx_data = mjx.put_data(mj_model, mj_data)

batch_data = jax.vmap(lambda _: mjx_data)(range(64))
# Batched model.geom_friction works for parallel domain randomization
fric_model = mjx_model.replace(geom_friction=jax.vmap(lambda _: mjx_model.geom_friction)(range(64)))
jit_step = jax.jit(jax.vmap(mjx.step, in_axes=(None, 0)))
batch_data = jit_step(fric_model, batch_data)
print(f"{fric_model.geom_friction.shape=}, {fric_model.body_pos.shape=}")
# Output: fric_model.geom_friction.shape=(64, 1, 3), fric_model.body_pos.shape=(2, 3)
```

Unfortunately, this does not work for all fields at the moment. For example, manually changing the mass (not caring for now if other attributes are set correctly as in https://github.com/google-deepmind/mujoco/issues/764#issuecomment-1464708759) will result in shape errors in `fwd_position`. 

```python
# Randomizing the mass in the same way does not work
mass_model = mjx_model.replace(body_mass=jax.vmap(lambda _: mjx_model.body_mass)(range(64)))
try:
    batch_data = jit_step(mass_model, batch_data)
except IndexError as e:
    print("Batching the body_mass property does not work, fails with: \n", e)
```

One can create full copies, change the masses, and `vmap` over both `Data` and `Model` in `mjx.step`, but this workflow is not described in the tutorial notebook and is more resource intensive. I think this should at least be documented somewhere, since the notebook in its current form suggests (though not explicitly mentioned) any field in `Model` can be batched and randomized.

```python
# We can randomize the mass property by batching the full mjx_model, but this will consume a lot
# more memory and requires a change in vmap in_axes.
batch_model = jax.vmap(lambda _: mjx_model)(range(64))
jit_step = jax.jit(jax.vmap(mjx.step))  # Note the change from in_axes=(None, 0) to in_axes=0
batch_data = jit_step(batch_model, batch_data)
print(f"{batch_model.body_mass.shape=}, {batch_model.body_pos.shape=}")
# Output: batch_model.body_mass.shape=(64, 2), batch_model.body_pos.shape=(64, 2, 3)
```

Long term, it would be great if we could batch individual fields of `Model` as is currently possible with `mjx.Model.geom_friction`. I'm not sure how much work that would be, but I'd be happy to help out if that is something you'd be interested in @btaba .

jonasweihing, 2025-02-13:
I am also running into this problem and have opened a [discussion](https://github.com/google-deepmind/mujoco/discussions/2406) as I had not found this issue previously.

I want to randomize the size of a geom (`geom_size`), which is possible batchable and therefore generally possible. However, changing the geom size requires a recompilation as the mass and therefore many other fields change. Some of these fields are batchable, but not all, such as `geom_rbound`, `dof_M0` and also all `stat` fields.
This means that it is not really possible to change the size of a geometry. 

I would really appreciate any progress in this direction to make randomizing model properties other than [visuals](https://github.com/google-deepmind/mujoco_playground/blob/main/mujoco_playground/_src/manipulation/franka_emika_panda/randomize_vision.py#L58) possible.
@btaba are there any plans for the future?

hshi74, 2025-02-13:
Not sure if this will solve your issue, but I implemented a workaround to randomize body mass in
[our MJX training code](https://github.com/hshi74/toddlerbot/blob/main/toddlerbot/locomotion/train_mjx.py#L182). Hope it helps!

amacati, 2025-02-13:
Thanks for posting this! I skipped through the code and it seems you require brax for your domain randomization. Is that correct, or can you also use the domain randomize code with pure MJX models?

hshi74, 2025-02-13:
Yes, brax is a necessary dependency for our codebase. I think extracting the domain randomization code from brax and applying it to pure MJX models should be possible unless I miss something important, it might take some engineering efforts though.

jonasweihing, 2025-02-14:
> I am also running into this problem and have opened a [discussion](https://github.com/google-deepmind/mujoco/discussions/2406) as I had not found this issue previously.
>
> I want to randomize the size [...]

I made a wrong assumption in my comment about the fields `geom_rbound`, `dof_M0` and `stat` not being batchable. Therefore, I wrote a small test to check which fields are not batched after a vmap. 

<details>
  <summary>Python code to test for batched fields</summary>

  ```Python
  import jax
  import mujoco
  import numpy as np
  from mujoco import mjx
  
  spec = mujoco.MjSpec()
  mj_model = spec.compile()
  model = mjx.put_model(mj_model)
  data = mjx.make_data(model)
  
  batch_size = 256
  
  data_batched = jax.vmap(lambda _: data)(range(batch_size))
  model_batched = jax.vmap(lambda _: model)(range(batch_size))
  
  non_batched_scalars = []
  non_batched_arrays = []
  
  def check_batched_fields(x, path=""):
      if isinstance(x, (jax.Array, np.ndarray, np.bool, np.int64)):
          if x.ndim == 0 or x.shape[0] != batch_size:
              non_batched_arrays.append(path)
      elif isinstance(x, (int, float, bool, str, bytes)):
          non_batched_scalars.append(path)
      elif isinstance(x, (tuple, list)):
          for idx, val in enumerate(x):
              check_batched_fields(val, f"{path}[{idx}]")
      elif hasattr(x, "__dict__"):
          for field in x.fields():
              current_path = f"{path}.{field.name}" if path else field.name
              check_batched_fields(getattr(x, field.name), current_path)
      else:
          print(f"Unexpected type {type(x)} of attribute {path}")
          
          
  check_batched_fields(model_batched, "")
  ```
</details>

JeyRunner, 2025-02-25:
@hshi74 interesting solution to the problem. Thanks for sharing your code! 
I was just thinking about the time it takes to precompute the mass attributes by calling mj_setConst sequentially. Is it sufficiently fast for e.g. 1000 envs (e.g. adding up 5s to the startup time does not matter I guess, but 1 minute would be pretty long)?

hshi74, 2025-02-25:
It's very fast! Based on my profiling results, calling `get_body_mass_attr_range` in my script takes 0.15s.