google-deepmind/mujoco #1610: Using mujoco(bin\compile.exe) to generate MJCF format XML files

xsc9202, 2024-04-22:
Hi,

I'm a  engineer and I'm trying to use MuJoCo(bin\bin\compile.exe) for the MJCF XML files. but encountered some problems:
we used Mujoco (mujoco-3.1.3-windows-x86_64\bin\compile.exe) to generate MJCF format XML files will result in losing the content of the first body.
I'm looking for some help:
1.Is it because we did not use the software correctly?
2.Or this is the mujoco software bug?
Thks!
[image]

yuvaltassa, 2024-04-22:
Are you converting from URDF?

xsc9202, 2024-04-23:
Hi
Yes , we used the URDF .

Best Regards!

外部邮件/External Mail

xsc9202, 2024-04-23:
Hi
we used the URDF and the STL files as following:

closed 2024-04-23 by xsc9202