google-deepmind/mujoco #1611: Issues with replaying mocap-recorded trajectories for object grasping

z-yf17, 2024-04-22:
Hello,

I'm encountering an issue when trying to replay mocap-recorded trajectories for grasping a target block with a robotic arm. Here's a detailed description of the process:

I set up a mocap at the wrist of the robotic arm to position it at the target block.
I controlled the gripper to close and grasp the block by setting the ctrl of the gripper.
I recorded the qpos of all the motors and the ctrl settings of the gripper during this operation.
I then initialized a new environment identical to the original and attempted to replay the recorded qpos and gripper ctrl values.
However, while the initial mocap control allows for successful grasping of the block, the replay causes the block to fly away upon contact. Could you please help me understand why this discrepancy occurs during the replay?
Videos are show above. My mujoco version is 200. Mujoco-py.

Thank you for your assistance.

https://github.com/google-deepmind/mujoco/assets/86708745/a4269c61-b0f8-45fd-9f8c-311f6d17e827

https://github.com/google-deepmind/mujoco/assets/86708745/f33413ca-78f6-422f-822c-dadf5250acf6

z-yf17, 2024-04-22:
[franka_ee_grasp.zip](https://github.com/google-deepmind/mujoco/files/15058388/franka_ee_grasp.zip)
This zip consists of 2 .xml files, corresponding to mocap-control and joint-control. The only difference is the weld.

yuvaltassa, 2024-04-22:
you can't just replay qpos values like that. you are basically teleporting the arm, it only looks smooth because the jumps are small. if you want to replay this trajectory you need to replay the movement of the mocap body and let it pull on the arm in the same way.

z-yf17, 2024-04-22:
> you can't just replay qpos values like that. you are basically teleporting the arm, it only looks smooth because the jumps are small. if you want to replay this trajectory you need to replay the movement of the mocap body and let it pull on the arm in the same way.

You are right. When I replay the trajectory, I should control ctrl but not qpos. The problem is solved now, thanks!

closed 2024-04-22 by yuvaltassa