google-deepmind/mujoco #1614: Failed to find a finite solution：LQR for a two wheel self-balancing robot

15906185391, 2024-04-23:
Hi,

I'm a student and I'm trying to use MuJoCo for LQR control for a two wheel self-balancing robot. I follow the LQR tutorial but I cannot find a finite solution. 

It says "numpy.linalg.LinAlgError: Failed to find a finite solution."

Here is my robot model :

```XML
<mujoco model="two_wheel_robot">
  <compiler angle="radian" coordinate='local' inertiafromgeom='true'/>

  <asset>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
  </asset>

  <worldbody>
    <geom name="floor" size="0 0 .05" type="plane" material="grid" condim="3"/>
    <light name="spotlight" mode="targetbodycom" target="box" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 -6 4" cutoff="30"/>
    <body name="robot_body">
      <freejoint name="root"/>
      <body name="wheel" pos="0 0 0" quat="1 0 0 0">
        <body name="left_wheel" pos="0 -0.17035 0.05" quat="1 -1 0 0">
            <joint name="joint_left_wheel" axis="0 0 -1" damping="0.3"/>
            <geom name="left_wheel" type="cylinder" size="0.05 0.0125" pos="0 0 0.0125"/>
        </body>
        <body name="right_wheel" pos="0 0.17035 0.05" quat="1 -1 0 0">
            <joint name="joint_right_wheel" axis="0 0 -1" damping="0.3"/>
            <geom name="right_wheel" type="cylinder" size="0.05 0.0125" pos="0 0 -0.0125"/>
        </body>
      </body>
      <body name="box" pos="0 0 0.11" quat="1 0 0 0">
        <geom name="box" type="box" size="0.03 0.1353 0.08" pos="0 0 0"/>
      </body>
    </body>
  </worldbody>

  <actuator>
        <motor name="right_wheel" gear="40" joint="joint_right_wheel" />
        <motor name="left_wheel" gear="40" joint="joint_left_wheel"/>
    </actuator>

</mujoco>
```

Here is my code:
```python
import mujoco
import numpy as np
import scipy
import time
import mujoco.viewer
import sys

model = mujoco.MjModel.from_xml_path("two_wheel_robot.xml")
data = mujoco.MjData(model)
nv=model.nv
nu=model.nu

R = np.eye(nu)

mujoco.mj_resetData(model,data)
mujoco.mj_forward(model,data)
jac_com=np.zeros((3,nv))
mujoco.mj_jacSubtreeCom(model,data,jac_com,model.body('robot_body').id)

jac_wheel_com=np.zeros((3,nv))
mujoco.mj_jacSubtreeCom(model,data,jac_wheel_com,model.body('wheel').id)

jac_diff=jac_com-jac_wheel_com
Qbalance=jac_diff.T@jac_diff
joint_names=[model.joint(i).name for i in range(model.njnt)]
root_dofs=range(6)
balance_dofs = [
    model.joint(name).dofadr[0]
    for name in joint_names
    if 'wheel' in name
]

BALANCE_COST=1000
BALANCE_JOINT_COST=3
Qjoint=np.eye(nv)
Qjoint[root_dofs,root_dofs]*=0
Qjoint[balance_dofs,balance_dofs]*=BALANCE_JOINT_COST
Qpos=BALANCE_COST*Qbalance+Qjoint
Q=np.block([[Qpos,np.zeros((nv,nv))],
            [np.zeros((nv,2*nv))]])

ctrl0=data.ctrl
qpos0=data.qpos

A = np.zeros((2*nv, 2*nv))
B = np.zeros((2*nv, nu))
epsilon = 1e-6
flg_centered = True
mujoco.mjd_transitionFD(model, data, epsilon, flg_centered, A, B, None, None)

# Solve discrete Riccati equation.
P = scipy.linalg.solve_discrete_are(A, B, Q, R)
# Compute the feedback gain matrix K.
K = np.linalg.inv(R + B.T @ P @ B) @ B.T @ P @ A
```
I do not know whether A,B is right, or I choose wrong Q. Could you please help me?

yuvaltassa, 2024-04-23:
Can you print the matrix `Qqpos` and post here please?

15906185391, 2024-04-23:
> Can you print the matrix and post here please?`Qqpos`

Thanks for the reply. qpos0 is: [0. 0. 0. 1. 0. 0. 0. 0. 0.]

yuvaltassa, 2024-04-23:
That's not what I asked for.

15906185391, 2024-04-23:
> That's not what I asked for.

```bash
Qpos is: 
[[ 4.93038066e-29  0.00000000e+00  0.00000000e+00  0.00000000e+00
  -9.64431509e-15  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  4.93038066e-29  0.00000000e+00  9.64431509e-15
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  4.93038066e-29  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  9.64431509e-15  0.00000000e+00  1.88652398e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [-9.64431509e-15  0.00000000e+00  0.00000000e+00  0.00000000e+00
   1.88652398e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  3.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00
   0.00000000e+00  0.00000000e+00  0.00000000e+00  3.00000000e+00]]
```
I'm sorry I misunderstood you.

15906185391, 2024-04-24:
> That's not what I asked for.

I changes Q matrix many times, but still can't find a finite solution

yuvaltassa, 2024-04-25:
Can you print `B[nv:nv+6, :]`?

Use the legible printout command from the import cell in the colab please. Your previous reply was unreadable.

15906185391, 2024-04-26:
> Can you print `B[nv:nv+6, :]`?
> 
> Use the legible printout command from the import cell in the colab please. Your previous reply was unreadable.

```python
B[nv:nv+6, :] is:
 [[-0.3   -0.3  ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 2.941  2.941]
 [ 0.     0.   ]]
```
Thanks for reply. I set the output legible. Besides,
```python
Qpos is:
 [[ 0.  0.  0.  0. -0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0. 36.  0.  0.  0.  0.]
 [-0.  0.  0.  0. 36.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  3.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  3.]]
```
```python
Q is:
[[ 0.  0.  0.  0. -0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0. 36.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [-0.  0.  0.  0. 36.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  3.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  3.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]
 [ 0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.  0.]]
```
```python
A is:
 [[     1.         0.         0.         0.        -0.         0.         0.         0.         0.002      0.         0.         0.         0.         0.         0.         0.   ]
 [     0.         1.         0.        -4.265     -0.         0.         0.         0.         0.         0.002      0.         0.         0.         0.         0.         0.   ]
 [     0.         0.       -17.054      0.        -0.         0.         0.         0.         0.         0.         0.002      0.         0.         0.         0.         0.   ]
 [     0.         0.         0.      -132.613     -0.         0.         0.         0.         0.         0.         0.         0.002      0.         0.         0.         0.   ]
 [     0.         0.         0.         0.         1.001      0.         0.         0.         0.         0.         0.         0.         0.002      0.        -0.        -0.   ]
 [     0.         0.         0.         0.         0.         1.         0.         0.         0.         0.         0.         0.         0.         0.002     -0.        -0.   ]
 [     0.         0.         0.         0.         0.        -0.         1.         0.         0.         0.         0.         0.         0.         0.         0.001     -0.   ]
 [     0.         0.         0.         0.         0.         0.         0.         1.         0.         0.         0.         0.         0.         0.        -0.         0.001]
 [     0.         0.         0.         0.        -0.032      0.         0.         0.         1.         0.         0.         0.         0.         0.         0.002      0.002]
 [     0.        -0.         0.     -2132.487     -0.         0.         0.         0.         0.         1.         0.         0.         0.         0.         0.         0.   ]
 [     0.         0.     -9026.979      0.        -0.         0.         0.         0.         0.         0.         1.         0.         0.         0.         0.         0.   ]
 [     0.        -0.         0.    -66806.536     -0.         0.         0.         0.         0.         0.         0.         1.         0.         0.         0.         0.   ]
 [     0.         0.         0.         0.         0.319     -0.         0.         0.         0.         0.         0.         0.         1.         0.        -0.022     -0.022]
 [     0.         0.         0.         0.         0.         0.         0.         0.         0.         0.         0.         0.         0.         1.        -0.        -0.   ]
 [     0.         0.         0.         0.         0.191     -0.         0.         0.         0.         0.         0.         0.         0.         0.         0.284     -0.006]
 [     0.         0.         0.         0.         0.191      0.         0.         0.         0.         0.         0.         0.         0.         0.        -0.006      0.284]]
```
```python
B is:
 [[-0.001 -0.001]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.006  0.006]
 [ 0.     0.   ]
 [ 0.002  0.191]
 [ 0.191  0.002]
 [-0.3   -0.3  ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 2.941  2.941]
 [ 0.     0.   ]
 [ 0.854 95.479]
 [95.479  0.854]]
```
Thank you very much for taking the time to help solve the problem!

yuvaltassa, 2024-04-26:
Can you please add, right after your call to `mj_resetData`, the below, and let me know if anything changes?
```py
for _ range(10):
  mujoco.mj_step(model, data)
```

15906185391, 2024-04-26:
> Can you please add, right after your call to `mj_resetData`, the below, and let me know if anything changes?
> 
> ```python
> for _ range(10):
>   mujoco.mj_step(model, data)
> ```

Yes, A and B change a lot. Now
```python
A is:
 [[ 1.     0.     0.     0.    -0.     0.     0.     0.     0.002  0.     0.     0.    -0.    -0.     0.     0.   ]
 [ 0.     1.    -0.    -0.    -0.    -0.     0.     0.     0.     0.002  0.    -0.     0.    -0.     0.     0.   ]
 [ 0.     0.     0.989  0.     0.     0.     0.     0.     0.     0.     0.002 -0.     0.    -0.     0.     0.   ]
 [ 0.     0.     0.     0.989 -0.    -0.     0.     0.     0.    -0.     0.     0.002  0.    -0.     0.     0.   ]
 [ 0.     0.     0.     0.     1.001  0.     0.     0.     0.001  0.     0.     0.     0.002  0.    -0.    -0.   ]
 [ 0.     0.     0.     0.    -0.     1.     0.     0.    -0.     0.     0.     0.     0.     0.002  0.    -0.   ]
 [ 0.     0.     0.     0.     0.    -0.     1.     0.    -0.002  0.     0.     0.    -0.    -0.     0.002 -0.   ]
 [ 0.     0.     0.     0.     0.    -0.     0.     1.    -0.002  0.     0.     0.    -0.     0.    -0.     0.002]
 [ 0.     0.     0.     0.    -0.036  0.     0.     0.     0.947  0.     0.     0.    -0.    -0.     0.015  0.015]
 [ 0.    -0.    -0.    -0.081 -0.    -0.     0.     0.     0.     0.818  0.    -0.003  0.    -0.     0.     0.   ]
 [ 0.     0.    -5.532  0.     0.     0.     0.     0.     0.     0.     0.805 -0.     0.    -0.     0.     0.   ]
 [ 0.    -0.    -0.    -5.614 -0.    -0.     0.     0.     0.    -0.057  0.     0.802  0.    -0.     0.     0.   ]
 [ 0.     0.     0.     0.     0.36  -0.     0.     0.     0.401  0.     0.     0.     1.     0.    -0.12  -0.12 ]
 [ 0.     0.     0.     0.    -0.     0.     0.     0.    -0.     0.     0.     0.     0.     0.953  0.05  -0.05 ]
 [ 0.     0.     0.     0.     0.216 -0.     0.     0.    -0.951  0.     0.     0.    -0.    -0.148  0.792 -0.049]
 [ 0.     0.     0.     0.     0.216 -0.     0.     0.    -0.951  0.     0.     0.    -0.     0.148 -0.049  0.792]]
```
```python
B is:
 [[-0.004 -0.004]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.035  0.035]
 [ 0.015 -0.015]
 [ 0.014  0.042]
 [ 0.042  0.014]
 [-2.21  -2.21 ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [17.374 17.374]
 [ 7.28  -7.28 ]
 [ 6.813 21.056]
 [21.056  6.813]]
```
```python
B[nv:nv+6, :] is:
 [[-2.21  -2.21 ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [ 0.     0.   ]
 [17.374 17.374]
 [ 7.28  -7.28 ]]
```
Thanks so much! I am trying to change Q to find a solution.

yuvaltassa, 2024-04-26:
Do you understand what happened? There is a whole section in the colab about getting the contacts correct at the initial pose. It is highly related to the problem you had with your model. It was basically floating in the air (at height 0, to numerical precision) with no proper contact, so the derivatives didn't show that applying torque to the wheels would have an effect on the free joint...

closed 2024-04-26 by yuvaltassa

15906185391, 2024-04-30:
> Do you understand what happened? There is a whole section in the colab about getting the contacts correct at the initial pose. It is highly related to the problem you had with your model. It was [...]

Thank you very much for your patience in replying and explaining, I am following the documentation on the official website to learn how to model the robot correctly and rationally.
Besides, I would like to ask you a question, the matrices A and B I got are partially correct now. My self-balancing cart can only go forward and backward at the moment, and it can't turn, I think it's still a modeling problem, but I don't know how to improve it, Do you know where I should modify the model?
Happy Labor Day!

mamoruoka, 2024-11-20:
@15906185391 
Please refer to the following code.
I have changed the qpos[2], jac_wheel_com(fixed to use mujoco.mj_jacBodyCom) and BALANCE_COST.

```python
model = mujoco.MjModel.from_xml_path(""two_wheel_robot.xmll")
data = mujoco.MjData(model)
mujoco.mj_forward(model, data)
data.qacc = 0  # Assert that there is no the acceleration.
data.qpos[2] += -0.000304
qpos0 = data.qpos.copy()  # Save the position setpoint.
mujoco.mj_inverse(model, data)
qfrc0 = data.qfrc_inverse.copy()

actuator_moment = data.actuator_moment
ctrl0 = np.atleast_2d(qfrc0) @ np.linalg.pinv(actuator_moment)
ctrl0 = ctrl0.flatten()

nv=model.nv
nu=model.nu

R = np.eye(nu)

mujoco.mj_resetData(model, data)
data.qpos = qpos0
mujoco.mj_forward(model, data)
jac_com=np.zeros((3,nv))
mujoco.mj_jacSubtreeCom(model,data,jac_com,model.body('robot_body').id)

jac_wheel_com=np.zeros((3,nv))
mujoco.mj_jacBodyCom(model,data,jac_wheel_com,None,model.body('wheel').id)

jac_diff=jac_com-jac_wheel_com
Qbalance=jac_diff.T@jac_diff
joint_names=[model.joint(i).name for i in range(model.njnt)]
root_dofs=range(6)
balance_dofs = [
    model.joint(name).dofadr[0]
    for name in joint_names
    if 'wheel' in name
]

BALANCE_COST = 1000000000  # Balancing.
BALANCE_JOINT_COST=3
Qjoint=np.eye(nv)
Qjoint[root_dofs,root_dofs]*=0
Qjoint[balance_dofs,balance_dofs]*=BALANCE_JOINT_COST
Qpos=BALANCE_COST*Qbalance+Qjoint
Q=np.block([[Qpos,np.zeros((nv,nv))],
            [np.zeros((nv,2*nv))]])

# Set the initial state and control.
mujoco.mj_resetData(model, data)
data.ctrl = ctrl0
data.qpos = qpos0

A = np.zeros((2*nv, 2*nv))
B = np.zeros((2*nv, nu))
epsilon = 1e-6
flg_centered = True
mujoco.mjd_transitionFD(model, data, epsilon, flg_centered, A, B, None, None)

# Solve discrete Riccati equation.
P = scipy.linalg.solve_discrete_are(A, B, Q, R)
# Compute the feedback gain matrix K.
K = np.linalg.inv(R + B.T @ P @ B) @ B.T @ P @ A

# Reset data, set initial pose.
mujoco.mj_resetData(model, data)
data.qpos = qpos0

# Allocate position difference dq.
dq = np.zeros(model.nv)

with mujoco.viewer.launch_passive(model, data) as viewer:
    start = time.time()
    data.ctrl = ctrl0

    while viewer.is_running():
        step_start = time.time()

        # Get state difference dx.
        mujoco.mj_differentiatePos(model, dq, 1, qpos0, data.qpos)
        dx = np.hstack((dq, data.qvel)).T
        # LQR control law.
        data.ctrl = ctrl0 - K @ dx

        mujoco.mj_step(model, data)
        viewer.sync()

        time_until_next_step = model.opt.timestep - (time.time() - step_start)
        if time_until_next_step > 0:
            time.sleep(time_until_next_step)

```