google-deepmind/mujoco #1616: NaN problems in mjx speed benchmarks on cpu

fabinsch, 2024-04-23:
Hello guys and thanks for maintaining this library. I tried to run the `testspeed.py` script and encounter nans.

I am on Ubuntu 22.04 and I create a clean env and installed mujoco-mjx by 
```
conda create --name mjx_test python=3.10
pip install mujoco-mjx
```
then I launch the benchmarks by
```
python testspeed.py --mjcf humanoid/humanoid.xml
```
and I see the following
```
(mjx_test) ➜  mjx python testspeed.py --mjcf humanoid/humanoid.xml
Rolling out 1000 steps at dt = 0.005...
I0423 16:03:29.898261 140143597762368 xla_bridge.py:863] Unable to initialize backend 'cuda': 
I0423 16:03:29.898374 140143597762368 xla_bridge.py:863] Unable to initialize backend 'rocm': module 'jaxlib.xla_extension' has no attribute 'GpuAllocatorConfig'
I0423 16:03:29.898871 140143597762368 xla_bridge.py:863] Unable to initialize backend 'tpu': INTERNAL: Failed to open libtpu.so: libtpu.so: cannot open shared object file: No such file or directory
W0423 16:03:29.898980 140143597762368 xla_bridge.py:901] An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not installed. Falling back to cpu.
2024-04-23 16:03:53.934908: W external/xla/xla/service/cpu/onednn_matmul.cc:293] [Perf]: MatMul reference implementation being executed
2024-04-23 16:03:53.993505: W external/xla/xla/service/cpu/onednn_matmul.cc:293] [Perf]: MatMul reference implementation being executed
....
2024-04-23 16:04:45.988970: W external/xla/xla/service/cpu/onednn_matmul.cc:293] [Perf]: MatMul reference implementation being executed
result.qpos: [[[nan nan nan ... nan nan nan]
  [nan nan nan ... nan nan nan]
  [nan nan nan ... nan nan nan]
  ...
  [nan nan nan ... nan nan nan]
  [nan nan nan ... nan nan nan]
  [nan nan nan ... nan nan nan]]]

Summary for 1024 parallel rollouts

 Total JIT time: 23.58 s
 Total simulation time: 52.14 s
 Total steps per second: 19640
 Total realtime factor: 98.20 x
 Total time per step: 50.92 µs

```
After having added a line `print(f"result.qpos: {result.qpos}")` [here](https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/_src/test_util.py#L46).
The output of pip list is
```
(mjx_test) ➜  mjx pip list
Package             Version
------------------- --------
absl-py             2.1.0
etils               1.7.0
fsspec              2024.3.1
glfw                2.7.0
importlib_resources 6.4.0
jax                 0.4.26
jaxlib              0.4.26
ml-dtypes           0.4.0
mujoco              3.1.4
mujoco-mjx          3.1.4
numpy               1.26.4
opt-einsum          3.3.0
pip                 24.0
PyOpenGL            3.1.7
scipy               1.13.0
setuptools          69.5.1
trimesh             4.3.1
typing_extensions   4.11.0
wheel               0.43.0
zipp                3.18.1
```

I came across an issue mentioning the `MatMul reference implementation being executed` [here](https://github.com/google/jax/issues/19885). I tried to run the benchmarks on a mac m1 and I do not have the `MatMul reference` or nan issue anymore.

If you have any idea what is going on for mjx on ubuntu (CPU) I would be happy,
Thanks for your time

btaba, 2024-09-09:
Hi @fabinsch I just tried this on py 3.10 on CPU on MJ 3.2.2 and I can't repro the NaN issue. Are you still seeing this issue?

fabinsch, 2024-09-10:
Hey @btaba, thank you for looking into it. I just tried with the latest `main` and no longer have the problem.

closed 2024-09-10 by fabinsch