google-deepmind/mujoco #1617: Fume visualization in MuJoCo

karthyyy, 2024-04-24:
Hi,

  Is it possible to create fume/smoke visualization in MuJoCo for engines/thrusters ? I understand MuJoCo's primary purpose is not visualization but it will be interesting to see thrusters in action.

Regards
   Karthik

Balint-H, 2024-04-24:
Are you only after visualization of fumes during a physics simulation? If you really want to use MuJoCo for your physics then the Unity plugin would be your best bet, plenty of tutorials online to generate particle streams with Unity, and you can use MuJoCo to handle the physics.

karthyyy, 2024-04-25:
Hi @Balint-H ,

  Thanks for the reply. Yes, I am only visualizing the fumes. The system modelling involved good amount of dynamics and control. So MuJoCo is essential. I had seen some examples of using Unity as visual engine and MuJoCo as physics engine. Was searching for options to implement the fume visualization alone in MuJoCo. Thanks for the feedback, I will start with interfacing Unity. Saw your interface tutorial with Unity-ML. I will go through the same. Whether the canon simulation mentioned in the reply is available ?

Regards
   Karthik R

Balint-H, 2024-04-26:
https://github.com/google-deepmind/mujoco/assets/41113387/bd470813-08d7-4d96-a801-9fc2e19bb5ce

Here's a video of me using a smoke puff visual flair with the Unity plugin (to stress test a locomotion policy in VR). There's nothing advanced with generating the smoke, it is just a Particle System component being triggered at the same the MuJoCo simulated projectile is launched.

Balint-H, 2024-04-26:
https://github.com/google-deepmind/mujoco/assets/41113387/2866fd9d-e613-4475-a1ae-e13610048efc

Here's a clip of me using a Particle System component from Unity for a visual flair while stress-testing a locomotion policy. There's nothing advanced abouth the smoke generation, it is triggered to generate every time the mujoco simulated projectile is launch. You could controle parameters of a particle generator like this in script to synchronise it with your thrust.

karthyyy, 2024-04-30:
@Balint-H ,

  Thanks a lot. The demo seems nice. :)...

   Sure. I will go through the tutorial and come back with something.

Regards
   Karthik R

closed 2024-04-30 by karthyyy