google-deepmind/mujoco #1621: Simulation time not proportional to simulation step number

MasterXiong, 2024-04-25:
Hi,

I'm a student and I'm trying to use MuJoCo for reinforcement learning on robotics. I'm trying to accelerate my simulation code by using a larger `timestep`. Specifically, my current code has a timestep of 0.005s, and for each step in the MDP, the control action will be executed for four times in Mujoco simulation, before a new control action is generated for the next step in the MDP. I'm trying to accelerate it by changing the Mujoco timestep to 0.02s, and only calling the Mujoco simulation for one time in each MDP step, which I expect to give me a speedup of 4 times. However, empirically I found that speedup is just about 1.5 times instead of 4. I was wondering that if I've got something wrong here? Or is it a normal phenomenon in Mujoco? Thanks for your help!

yuvaltassa, 2024-04-25:
Yes, this makes sense for a certain class of models. The constraint solver is sometimes significantly does up by warmstarting with the previous solution. This speedup effect is larger for smaller timesteps, since in that case subsequent states are more similar (fewer changes in contact state, etc.).

closed 2024-04-25 by yuvaltassa

MasterXiong, 2024-04-25:
@yuvaltassa Thanks for your reply! I was wondering if there are some other common ways to accelerate Mujoco simulation? Or is there any reference for this? Thanks!

yuvaltassa, 2024-04-27:
https://mujoco.readthedocs.io/en/latest/modeling.html#performance-tuning