google-deepmind/mujoco #1624: How to route a cable (over a pulley)

danachee, 2024-04-26:
Hi,

I'm a engineer that is very new to this space and I'm trying to use MuJoCo for a simulation project.

I'm looking for some help with connecting both ends of a cable.  One to the top of a beam (with a slider joint) and one to its payload.

Here is a model which explains my question:

<details>
The goal is to have a cable that is attached to a beam (at the top) and drapes over the end (where there would be a pulley).  When the slider moves back, the payload rises.  When the slider moves forward, the payload goes down.   First problem, I cannot figure out how to get the end of the cable to attach to the slider "hook" (green box).  And I would also like to make it ride along the beam (grey box) and drape off of the short edge.  Imagine a fishing rod with a line going off the end with a weight on it.

Anyway, google hasn't turned up anyone who has connected a cable this way.   Any hints??

Thanks!

```XML
<mujoco model="CableSlide">
  <include file="scene.xml"/>

  <extension>
    <plugin plugin="mujoco.elasticity.cable"/>
  </extension>

  <worldbody>
    <body name="payload">
      <geom type="box" size=".05 .05 .05" rgba=".5 .2 .8 1" mass="50"/>
      <freejoint/>
      <composite type="cable" curve="s" count="50" initial="ball">
        <pin coord="0"/>
        <joint kind="main" damping=".015"/>
        <geom type="capsule" size=".005" rgba=".8 .2 .1 1" condim="1"/>
      </composite>
    </body>
    <body name="beam" pos="0 0 .6">
      <geom type="box" size=".45 .0375 .0375" mass="100"/>
      <body name="slider" pos="0 .01 .05">
        <geom type="box" size=".01 .01 .01" rgba="0 1 0 1" mass="10"/>
        <joint name="jslider" type="slide" axis="1 0 0" damping=".1" limited="true" range="-.25 .25"/>
      </body>
    </body>
  </worldbody>

  <equality>
    <!-- <weld name="w0" body1="B_last" body2="slider" anchor="0 0 0"/> -->
    <connect name="c1" body1="slider" body2="B_last" anchor="0 0 0"/>
  </equality>
  <contact>
    <exclude body1="slider" body2="B_last"/>
  </contact>
  <actuator>
    <velocity name="slider" joint="jslider"/>
  </actuator>
</mujoco>
```

</details>

Here is a screenshot illustrating my question:
[image]

Balint-H, 2024-04-26:
Hello, could you answer a couple questions?

Does the fact that the cable can bend and twist at any point important, or would it be fine if it was piece-wise straight and only bent at the pulley?

Is the payload supposed to be able to be pulled on top of the beam, or only lifted up in the air?

Depending on the answers it may be more appropriate to simulate this system with tendons instead of cables.

danachee, 2024-04-26:
Well, it would act like a fishing pole and string, so it would only bend as it goes over the edge of the rod, but it will then swing and spin around with the payload.

The payload will never be on top of the beam, just to the edge.

Is it true that tendons can stretch?  In this case, like with a regular line/rope, it should maintain its length and not stretch or shrink.

Thanks!

Balint-H, 2024-04-26:
Tendons can be stretchable, but you can just as well define rigid, or rope-like tendons by defining a max length they can be. You can define geoms that a tendon must wrap around if its path would cross it as well.

danachee, 2024-04-26:
One other question, do tendons work with MuJoCo XLA (mjx)?  That's the package we need to use.

Thanks for helping!  And I'm guessing cables are not designed (yet?) to be routeable?

danachee, 2024-05-01:
Do you have a rough ETA for the tendons?  Weeks, months, ...?

Thanks!

mihirp1998, 2024-07-20:
Hi,

I wanted to follow up and ask how could one go about doing this via a cable? I dont want tendons as they can change their lenghts and would really prefer a pulley instead.

yuvaltassa, 2024-07-20:
Tendons are coming to MJX!
Fixed tendons next week, spatial tendons soon thereafter. cc @erikfrey 🙂

I would recommend using tendons with length constraints, unless performance doesn't matter to you at all. Cables are expensive!

mihirp1998, 2024-07-20:
[image]

Thanks for the reply.

I'm trying to simulate questions like these from physics textbooks (also i dont need mjx), where it's requirement that the length of the string doesn't change, however  if i understand correctly it's not possible to entirely fix the length of a tendon, as when i keep the range limits to be the same value, it doesn't work.

I don't have any speed issues right now, so i was thinking of going about with cable. However not sure if there is an easy way to route cable over a pulley

yuvaltassa, 2024-07-20:
You can make the tendon length limit as stiff as you like...

The model mentioned in this video is an example

https://youtu.be/I2q7D0Vda-A?si=Tz8mDu9s6z4zt5Vl

mihirp1998, 2024-07-20:
It's mainly like this right? 

`limited="true" range="0.5 0.5000000000000001"`

vldzrcky, 2024-07-26:
Hi,

I'm currently using cables in a similar manner in a crane model, but it is very slow and unstable.  I wanted to see if tendons are a good alternative.  I have some questions about tendons.

Is there a way to enable collisions with a tendon?  Currently the tendon is passing through objects that are in its way instead of colliding with them.

If there is a way to enable collisions, is it also possible to make a tendon wrap around other object that it collides with, similar to a cable?