google-deepmind/mujoco #1626: Creating a bright LED like object in Mujoco ?

dosssman, 2024-04-30:
Hello everyone.
Thank you for your continued work on this library.

I am currently working with Mujoco as a wrapper over a real world Franka robotic arm.
More specifically, I am trying to create an object looking like a moderately bright LED.
For example, the blue status LED at the base of the Franka robotics Arm (picture below). 

So far I have played with the `light` from the [docs](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-light) but it seems to only support general room lighting.

Is perhaps another way to achieve the aforementioned desired effect ?
Or more generally, an object that is emitting light ? 

Thank you in advance for your time.
Looking forward to hear from you.
Best regards.

[image]

Balint-H, 2024-05-02:
Check out the [emission attribute of the material element description](https://mujoco.readthedocs.io/en/stable/XMLreference.html#asset-material). You can assign it to a geom to have a bright object in scene. To turn it on or off, you could set the LED geom in a group of its own and disable/enable it through the visual options (check out the tutorial Colab to see how you can set visualization options). Alternatively you could change the material/color of the LED geom if you want to toggle it. 

This should be enough for simple visualization/state feedback.

dosssman, 2024-05-07:
Thank you for your insightful answer !
I managed to get it working using `<material name="EmissionObject" emission="1" [...] >` and then `<geom material="EmissionObject" [...] />` as suggested.
Closing the issue now.

closed 2024-05-07 by dosssman