google-deepmind/mujoco #1627: segmentation fault when use nlopt_optimize of NLopt  in mjcb_control

songdaegeun, 2024-05-01:
Hi, I'm a student and I'm using mujoco3.1.4. with nlopt 2.7.1.

In my code, inverse_kinematics() is called in callback "mycontroller". 
then segmentation fault occur.

when I try to call mycontroller() in main(), It work well.

I don't understand why the result is different depending on where you call the callback function.
Is this a bug?

<debug result>
```
Process 90889 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=2, address=0x16f603d98)
    frame #0: 0x0000000100841aec libmujoco.3.1.4.dylib`mj_kinematics + 612
libmujoco.3.1.4.dylib`mj_kinematics:
->  0x100841aec <+612>: str    x19, [sp, #0x8]
    0x100841af0 <+616>: add    x8, sp, #0x50
    0x100841af4 <+620>: add    x21, x8, #0xd
    0x100841af8 <+624>: adrp   x0, 444
Target 0: (test) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=2, address=0x16f603d98)
  * frame #0: 0x0000000100841aec libmujoco.3.1.4.dylib`mj_kinematics + 612
    frame #1: 0x0000000100961ac4 libmujoco.3.1.4.dylib`mj_fwdPosition + 136
    frame #2: 0x0000000100965bec libmujoco.3.1.4.dylib`mj_forwardSkip + 132
    frame #3: 0x0000000100002990 test`myequalityconstraints [inlined] simulator(Xin=<unavailable>, Xout=<unavailable>) at constrained.c:32:2 [opt]
    frame #4: 0x0000000100002950 test`myequalityconstraints(m=<unavailable>, result=0x000000011ea8bb80, n=<unavailable>, x=<unavailable>, grad=<unavailable>, equalitydata=<unavailable>) at constrained.c:5
```

<details>
  <summary>my code</summary>

test.c
```
#include<stdbool.h> //for bool
#include "mujoco/mujoco.h"
#include <GLFW/glfw3.h>
#include "stdio.h"
#include "stdlib.h"
#include "string.h"

#include "constrained.c"

double a = 0.25;
const double omega = 1;
double center_x = 1, center_z = 1.25;
double xref,zref,xact,zact;

// simulation end time
double simend = 2*3.14/omega+2;

//related to writing data to a file
FILE *fid;
int loop_index = 0;
const int data_chk_period = 50; //frequency at which data is written to a file

//Change the path 
char path[] = "./";
char datafile[] = "data.csv";
char xmlfile[] = "test.xml";

// MuJoCo data structures
mjModel* m = NULL;                  // MuJoCo model
mjData* d = NULL;                   // MuJoCo data
mjData* dsim = NULL; 
mjvCamera cam;                      // abstract camera
mjvOption opt;                      // visualization options
mjvScene scn;                       // abstract scene
mjrContext con;                     // custom GPU context
[... 434 lines cut ...]
	
	<body pos="0.5 0 1.25" euler="0 90 0">
		<geom type="cylinder" size=".05 .5" rgba="0 .9 0 1" mass="1"/>
		<joint name="pin" type="hinge" axis="0 -1 0" pos="0 0 -.5"/>
		<body pos="0 0.1 1" euler="0 0 0">
			<geom type="cylinder" size=".05 .5" rgba=".9 .9 .9 1" mass="1"/>
			<joint name="pin2" type="hinge" axis="0 -1 0" pos="0 0 -.5"/>
			<site name="tip" pos="0 0 .5" size="0.1"/>
		</body>	
	</body>
  </worldbody>
  <actuator>
	<position name="pservo1" joint="pin" kp="100" />
	<velocity name="vservo1" joint="pin" kv="10" />
	<position name="pservo2" joint="pin2" kp="100" />
	<velocity name="vservo2" joint="pin2" kv="10" />
  </actuator>
  <sensor>
	<framepos  objtype='site' objname='tip' />
	<framelinvel  objtype='site' objname='tip' />
  </sensor>

</mujoco>

```

yuvaltassa, 2024-05-01:
Nothing jumps out as obviously wrong, but of course staring at code is not the way to fix this. I'd recommend [building from source](https://mujoco.readthedocs.io/en/latest/programming/index.html#building-from-source) and using a debugger.

songdaegeun, 2024-05-02:
> Nothing jumps out as obviously wrong, but of course staring at code is not the way to fix this. I'd recommend [building from source](https://mujoco.readthedocs.io/en/latest/programming/index.html#building-from-source) and using a debugger.

thank you for your help! It was due to an infinite recursive call in the callback function.

closed 2024-05-02 by songdaegeun