google-deepmind/mujoco #1630: Soft Robot Simulation: (a) Example XML files, and (b) difference between Composite and Deformable soft objects

kurtenkera, 2024-05-02:
Hi,

I'm a PhD researcher and I'm trying to use MuJoCo for soft-robot simulation.

I have two questions:

(1) I'm looking for some example soft-robot XML files to learn how to model my own soft-robot in MuJoCo - are there any prominent examples? I can see in the [MuJoCo menageries](https://github.com/google-deepmind/mujoco_menagerie/tree/main) that there are several **rigid robot XML files** which seem to use [obj2mjcf](https://github.com/kevinzakka/obj2mjcf) to decompose an OBJ file of a rigid robot into various sub-meshes, which can then be used to create an XML file. Can something analogous be done for soft robots, but using the `flexcomp` element instead of the rigid `geom` element? This is hard for me to imagine, since by definition a soft-robot isn't a series of rigid links connected by various joints, so I don't see the need to generate several flexcomp sub-meshes for a soft-robot (unless I'm mistaken)?
(2) I'm struggling to understand the differences between [composite objects](https://mujoco.readthedocs.io/en/latest/modeling.html#ccomposite) and [deformable objects](https://mujoco.readthedocs.io/en/latest/modeling.html#cdeformable). From reading through these two links, it seems like deformable objects can more reliably model soft-bodies, but I'd like to understand the actual differences?

I appreciate any help/feedback/comments dramatically! Thanks in advance.

quagla, 2024-05-02:
Hi, the answer to (2) is clear: please use deformable objects, we will deprecate 2D/3D soft bodies from composite since they use only spheres at mesh vertices for collisions.

For (1), you have to think of `flexcomp` as a collection of `body`'s (one per mesh vertex), so it's not an alternative to `geom`. A deformable objects introduces 3*nvert degrees of freedom, while a geom introduces none. We have some examples here https://github.com/google-deepmind/mujoco/tree/main/model/plugin/elasticity I think the one you may want to look at is `trampoline_flex` since it shows how to pin a soft object to a body.

Let me know if you have more questions.

kurtenkera, 2024-05-07:
Hi @quagla, for `flexcomp` we reference 3D mesh files. So is the 'nvert' you're referring to the number of total nodes in the 3D mesh? Or only the total number of vertices included for collisions/in the convex hull of the 3D mesh?

quagla, 2024-05-07:
The total number. It's really the total number of independent deformation modes of the soft object.

closed 2024-05-07 by quagla

kurtenkera, 2024-05-07:
How do I obtain the vertices in the convex hull of a `flexcomp` object that references a 3D MSH file? I'm struggling to also obtain the vertices in the convex hull of (a) primitive `geom` objects and (b) rigid `geom` objects that reference STL files (as described [here](https://github.com/google-deepmind/mujoco/issues/1635)?