google-deepmind/mujoco #1631: Proposal for linear parametrization of inverse dynamics for SysID

lvjonok, 2024-05-03:
Hello MuJoCo Team,

I am currently finalizing my thesis in system identification (SysID) and adaptive control, together with @simeon-ned, we are working on SysID and adaptive control. We are very excited about the recent April update that introduced optimization and mentioned upcoming tutorials on SysID. Given our consistent use of MuJoCo for our research, we have developed a small [GitHub repository dedicated to SysID in MuJoCo](https://github.com/lvjonok/mujoco-sysid?tab=readme-ov-file) .

We have focused on the regressor representation, a linear parametrization of inverse dynamics, which is pivotal in SysID for robotic systems. To our knowledge, dedicated functions for this representation are not available in MuJoCo, prompting us to develop a prototype. Here are the main functions we introduced:
- [`mj_bodyRegressor`](https://github.com/lvjonok/mujoco-sysid/blob/1a041888600e9539dcad3f1cb8093132eac8d12f/mujoco_sysid/modeling.py#L56): This function calculates the regressor matrix for a single rigid body within a MuJoCo model.
- [`mj_jointRegressor`](https://github.com/lvjonok/mujoco-sysid/blob/1a041888600e9539dcad3f1cb8093132eac8d12f/mujoco_sysid/modeling.py#L91): This function calculates the regressor matrix for the entire inverse dynamics within a MuJoCo model.

**Demo and Examples**:

[We have also included a demo notebook showcasing SysID](https://colab.research.google.com/github/lvjonok/mujoco-sysid/blob/master/examples/mujoco_sysid_demo.ipynb) in robotic systems using the regressor functions. Current examples include:
- Estimation of cart-pole inertial parameters through random forcing and LQR stabilization.
- Identification of the end-effector load for the Franka Emika Panda and compensation using inverse dynamics.
- Determination of mass, center of mass, and spatial inertia for a Skydio X2 Quadrotor following trajectories via LTV LQR.

We plan to expand these examples to include more advanced techniques such as adaptive control, Riemannian regularization on parameter manifolds, physical consistency, and log-Cholesky inertia parametrization.

If you find this work useful and believe it deserves inclusion in the official MuJoCo engine and MJX, we would be delighted to contribute these capabilities further and assist in any way possible. Additionally, we are eager to enhance the demo notebook with more examples. We look forward to your feedback and hope to contribute meaningfully to the MuJoCo community. 

Lastly, we want to express our gratitude for MuJoCo. We work with it daily and find it to be an incredibly convenient, powerful, and versatile tool.

yuvaltassa, 2024-05-05:
Hi!

What is a "body regressor" and "joint regressor"? You say "the regressor matrix for a single rigid body", I don't know what that is, please define it.

simeon-ned, 2024-05-05:
Hello @yuvaltassa! Thanks for your reply!

In this context, a "regressor" refers to a matrix that linearly relates the inverse dynamics of a rigid body or a full mechanical system to its inertial parameters. These inertial parameters typically include the mass \( m \), the first moment of mass (denoted as \( m r_x, m r_y, m r_z \)), and the elements of the inertia matrix (\( I_{xx}, I_{xy}, I_{yy}, I_{xz}, I_{yz}, I_{zz} \)). We often represent these parameters as a vector, denoted by \( \theta \).

The equation for inverse dynamics can thus be expressed in the form:

\[ M(q) \dot{v} + h(q, v) = Y(q, v, \dot{v}) \theta = Q \]

This framework is extensively utilized in fields such as adaptive control and system identification. One may also define such regressors on other mechanical quantities like energy, momentum, and power, discussed further in [this paper](https://ieeexplore.ieee.org/document/574537).

We provide a short introduction to these concepts and list of related references in our [Colab notebook](https://colab.research.google.com/drive/1xFte2FT0nQ0ePs02BoOx4CmLLw5U-OUZ#scrollTo=V1cs1QQnRzWz). Additionally, the practical implementation of these concepts can be found in our repository within the [mujoco-sysid modeling.py](https://github.com/lvjonok/mujoco-sysid/blob/feat/regressor/mujoco_sysid/modeling.py).

yuvaltassa, 2024-05-05:
Okay, let me see if I follow. Given the position, velocity and acceleration of a body or an entire mechanism, you rewrite the dynamics 
$$M(q) \dot{v} + h(q, v) = \tau$$
as
$$Y(q, v, \dot{v}) \theta = \tau$$
Where the vector $\theta$ is effectively a rearrangement of the (unknown) inertia $M$, and $Y$ is the matrix that you compute. Given measurements of $q, v, \dot{v}$ and the computed matrix $Y$, the user can recover an estimate of the inertial parameters $\theta$.

Is this correct?

PS note that GH markdown supports LaTex 🙂

simeon-ned, 2024-05-06:
Exactly so, you've got it right. And depending on the parameterization of the vector $\theta$, we can formulate the parameter estimates as simple linear least squares or more general nonlinear ones (solved for example, with `mujoco.minimize`)

PS: Thanks for the point about LaTeX, I've modified my comment above.

lvjonok, 2024-05-06:
Moreover, it is not the only way to express parameterized dynamics linearly with respect to parameters. Energy and momentum of the system can be written linearly with respect to dynamical parameters too. 

They are classified as acceleration-free models, so you need just $q, v$ to formulate a regressor in form $A(q, v) \theta = b$.

chinahuangyong, 2024-05-06:
good job, i also have some similar work, but the simulator tool is matlab.

closed 2026-09-18 by yuvaltassa