google-deepmind/mujoco #1641: [Bug Report] Suspected bug list of MjBindings.cs

kshfire, 2024-05-10:
At first, thanks to the MjBindings.cs file, it is easy to use in C#.
In the meantime, I think there seem to be some bugs in the MjBindings.cs file.
The suspected bugs found are as follows (please check it)

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
**[return: MarshalAs(UnmanagedType.LPStr)]**
public static unsafe extern **string** mj_id2name(mjModel_* m, int type, int id);

=> I failed to use "mj_id2name" method, so I modified it as follows:

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
public static unsafe extern **IntPtr** mj_id2name(mjModel_* m, int type, int id);

**If string return is still OK, please let me know a sample code if possible. :)**

//------------------------------------------------------------------------------------

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
public static unsafe extern **mjuiThemeSpacing_*** mjui_themeSpacing(int ind);

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
public static unsafe extern **mjuiThemeColor_*** mjui_themeColor(int ind);

Looking at the original C code, the return types are not a pointer. Thus, these codes were modified as follows.

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
public static unsafe extern **mjuiThemeSpacing_** mjui_themeSpacing(int ind);

[DllImport("mujoco", CallingConvention = CallingConvention.Cdecl)]
public static unsafe extern **mjuiThemeColor_** mjui_themeColor(int ind);

I think it's a bug caused by a typo.

yuvaltassa, 2024-05-10:
cc @Balint-H

Balint-H, 2024-05-13:
I remember using id2name at some point in the past, but that it was quite finnicky, I'll have a look into it.

kshfire, 2024-05-13:
> I remember using id2name at some point in the past, but that it was quite finnicky, I'll have a look into it.

Below is the method I'm using (By googling)
**var ptr = mj_id2name(m_, (int)mjtObj.mjOBJ_BODY, selbody);**
string variable = **Marshal.PtrToStringAnsi(ptr);**

It would be best if we could use the return type of id2name as a C# string (if possible). :)