google-deepmind/mujoco #1642: Rotate and Move a Box Body w/ Hinge Joint

jonzamora, 2024-05-10:
Hi,

I'm a student researcher and I'm trying to use MuJoCo for creating a simple `SweepToTop` environment, inspired by [X-MAGICAL](https://github.com/kevinzakka/x-magical). In a simple 2D simulator, the task looks like this:

[image]

I'm recreating this task in MuJoCo, and i'm looking for some help with adding a Hinge Joint to this while preserving forward/backward left/right movement **after** rotating.

My current `scene.xml` is provided below:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="SweepToTop-v0">
    <visual>
        <headlight diffuse="0.8 0.8 0.8" ambient="0.4 0.4 0.4" specular="0.1 0.1 0.1" />
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1=".4 .6 .8" rgb2="0 0 0" height="800"
            width="800" mark="random" markrgb="1 1 1" />
        <texture name="grid" type="2d" builtin="checker" rgb1=".1 .1 .1" rgb2=".1 .1 .1"
            height="300" width="300" mark="edge" markrgb="0.8 0.8 0.8" />
        <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance="0.2" />
    </asset>

    <worldbody>
        <light name="light" pos="0 0 1" />
        <geom name="floor" type="plane" material="grid" size="2 2 .1" />
        <camera name="default" pos="0.000 -4.5 4.5"
            xyaxes="1.000 -0.000 0.000 0.000 0.707 0.707" />
        <camera name="top" pos="0.000 0.000 6.0"
            xyaxes="1.000 -0.000 0.000 0.000 0.707 0.000" />

        <!-- Environment Setup: Border Wall around the Grid -->
        <body name="top_wall" pos="0 2.1 0.2">
            <geom type="box" size="2 0.1 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="bottom_wall" pos="0 -2.1 0.2">
            <geom type="box" size="2 0.1 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="left_wall" pos="2.1 0 0.2">
            <geom type="box" size="0.1 2.2 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="right_wall" pos="-2.1 0 0.2">
            <geom type="box" size="0.1 2.2 0.2" rgba=".2 .2 .2 1" />
        </body>

        <!-- Agent: Shortstick -->
        <body name="shortstick" pos="0 0 0.1">
            <geom type="box" size="0.2 0.1 0.1" rgba="0.3 0.3 0.3 1" />
            <site name="left_eye_white" pos="-0.075 0.04 0.1" size="0.05 0.001 0.001" rgba="1 1 1 1" type="cylinder"></site>
            <site name="left_eye_black" pos="-0.075 0.06 0.1" size="0.025 0.002 0.001" rgba="0 0 0 1" type="cylinder"></site>
            <site name="right_eye_white" pos="0.075 0.04 0.1" size="0.05 0.001 0.001" rgba="1 1 1 1" type="cylinder"></site>
            <site name="right_eye_black" pos="0.075 0.06 0.1" size="0.025 0.002 0.001" rgba="0 0 0 1" type="cylinder"></site>
            <joint name="move_x" pos="0 0 0" axis="1 0 0" type="slide" />
            <joint name="move_y" pos="0 0 0" axis="0 1 0" type="slide" />
        </body>

        <body name="green_target" pos="0 1.8 0.1">
            <geom type="sphere" size="0.1" rgba="0 1 0 0.5" />
        </body>
    </worldbody>

    <!-- Define (x, y) action space with 2 actuators -->
    <actuator>
        <general name="move_x" joint="move_x" ctrlrange="-1 1" ctrllimited="true" gear="10" />
        <general name="move_y" joint="move_y" ctrlrange="-1 1" ctrllimited="true" gear="10" />
    </actuator>

    <keyframe>
        <key name="home" />
    </keyframe>
</mujoco>
```

</details>

Currently, my `shortstick` agent has 2 DOF with (x, y) action space (see the video i've attached). I've tried adding a hinge `joint` on the `z` axis of my shortstick `body`, but after I rotate the body, the `(x, y)` movement doesn't work like **before** I rotate the object. I've also tried changing the `ctrlrange="-1 1"` of the `(x, y)` joints to be more lenient, and I have also tried adding a range of (-180, 180) for the hinge joint, but the movement **after** I rotate the `shortstick` agent still doesn't work.

Any help here will be greatly appreciated, thanks!

https://github.com/google-deepmind/mujoco/assets/50164418/83972a06-cd94-4f69-afbc-f024c15cfec6

jonzamora, 2024-05-10:
For an example **with** a hinge joint added, see below:

<details>
  <summary>minimal XML w/ hinge joint</summary>
```
<mujoco model="SweepToTop-v0">
    <visual>
        <headlight diffuse="0.8 0.8 0.8" ambient="0.4 0.4 0.4" specular="0.1 0.1 0.1" />
    </visual>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1=".4 .6 .8" rgb2="0 0 0" height="800"
            width="800" mark="random" markrgb="1 1 1" />
        <texture name="grid" type="2d" builtin="checker" rgb1=".1 .1 .1" rgb2=".1 .1 .1"
            height="300" width="300" mark="edge" markrgb="0.8 0.8 0.8" />
        <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance="0.2" />
    </asset>

    <worldbody>
        <light name="light" pos="0 0 1" />
        <geom name="floor" type="plane" material="grid" size="2 2 .1" />
        <camera name="default" pos="0.000 -4.5 4.5"
            xyaxes="1.000 -0.000 0.000 0.000 0.707 0.707" />
        <camera name="top" pos="0.000 0.000 6.0"
            xyaxes="1.000 -0.000 0.000 0.000 0.707 0.000" />

        <!-- Environment Setup: Border Wall around the Grid -->
        <body name="top_wall" pos="0 2.1 0.2">
            <geom type="box" size="2 0.1 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="bottom_wall" pos="0 -2.1 0.2">
            <geom type="box" size="2 0.1 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="left_wall" pos="2.1 0 0.2">
            <geom type="box" size="0.1 2.2 0.2" rgba=".2 .2 .2 1" />
        </body>

        <body name="right_wall" pos="-2.1 0 0.2">
            <geom type="box" size="0.1 2.2 0.2" rgba=".2 .2 .2 1" />
        </body>

        <!-- Agent: Shortstick -->
        <body name="shortstick" pos="0 0 0.1">
            <geom type="box" size="0.2 0.1 0.1" rgba="0.3 0.3 0.3 1" />
            <site name="left_eye_white" pos="-0.075 0.04 0.1" size="0.05 0.001 0.001" rgba="1 1 1 1" type="cylinder"></site>
            <site name="left_eye_black" pos="-0.075 0.06 0.1" size="0.025 0.002 0.001" rgba="0 0 0 1" type="cylinder"></site>
            <site name="right_eye_white" pos="0.075 0.04 0.1" size="0.05 0.001 0.001" rgba="1 1 1 1" type="cylinder"></site>
            <site name="right_eye_black" pos="0.075 0.06 0.1" size="0.025 0.002 0.001" rgba="0 0 0 1" type="cylinder"></site>
            <joint name="move_x" pos="0 0 0" axis="1 0 0" type="slide" />
            <joint name="move_y" pos="0 0 0" axis="0 1 0" type="slide" />
            <joint name="rotate" type="hinge" pos="0 0 0" axis="0 0 1" range="-180 180" />
        </body>

        <body name="green_target" pos="0 1.8 0.1">
            <geom type="sphere" size="0.1" rgba="0 1 0 0.5" />
        </body>
    </worldbody>

    <!-- Define (x, y) action space with 2 actuators + Define (rotate) action -->
    <actuator>
        <general name="move_x" joint="move_x" ctrlrange="-1 1" ctrllimited="true" gear="10" />
        <general name="move_y" joint="move_y" ctrlrange="-1 1" ctrllimited="true" gear="10" />
        <general name="rotate" joint="rotate" ctrlrange="-1 1" ctrllimited="true" gear="10" />
    </actuator>

    <keyframe>
        <key name="home" />
    </keyframe>
</mujoco>
```
</details>

yuvaltassa, 2024-05-10:
The model looks fine, but the video doesn't show your problem, there is no rotation... Can you record another video and use the joint sliders instead of the actuator sliders (in Paused mode) to explain what the issue is?

jonzamora, 2024-05-10:
Hi @yuvaltassa , sure! Here's a video showing the issue

https://github.com/google-deepmind/mujoco/assets/50164418/ae05b2c8-267f-4308-acf3-51fdcd335956

As you can see, if I fully rotate the Box object, then try to apply `x` or `y` motion, it refuses to move.

Any ideas or suggestions why this could be?

jonzamora, 2024-05-10:
@yuvaltassa I tried it in Pause Mode w/ Joint Sliders like you suggested, here's the video of that:

https://github.com/google-deepmind/mujoco/assets/50164418/6646a250-6500-4bc2-846c-5e66a7a5a6ae

It looks like the Box is not traveling beyond the space of [1, 1] 😬

jonzamora, 2024-05-10:
@yuvaltassa I added `range="-1.8 1.8"` to the `move_x` and `move_y` joints to get complete coverage of the grid, but my issue with the Actuators still persists.

```XML
<joint name="move_x" pos="0 0 0" axis="1 0 0" type="slide" range="-1.8 1.8" />
<joint name="move_y" pos="0 0 0" axis="0 1 0" type="slide" range="-1.8 1.8" />
```

If you have any ideas about how to fix the Control actuators so they don't get "stuck" like in the above video, i'll appreciate any guidance!

yuvaltassa, 2024-05-10:
I don't think the issue is with the actuators. Did you visualize contacts / contact forces ("C" and "F" keys)?

I suspect your thing is getting stuck on the floor. Try 
```xml
<geom name="floor" type="plane" material="grid" size="2 2 .1" contype="0" conaffinity="0"/>
```

jonzamora, 2024-05-10:
@yuvaltassa This fixed my issue, thank you! I repeated the same test as above (rotate the object, then try to move in `x` and `y` direction with actuators) and it works how I want 🥳

https://github.com/google-deepmind/mujoco/assets/50164418/379f4e99-a2f2-4182-8ba3-007b9e903b14

closed 2024-05-10 by jonzamora