google-deepmind/mujoco #1645: Issue with Simultaneous Joint Movements in Telescopic Arm Not Reaching Target Positions

kankanzheli, 2024-05-10:
Dear MuJoCo Support Team,
﻿
I am currently developing a project using MuJoCo and have encountered an issue with the control of multiple joints in a telescopic arm segment of my robot. I am seeking your advice on how to address this problem effectively.
﻿
Issue Description:
In my robotic model, there are several joints, including those for rotation and translation movements, which all function as expected. However, I am facing a specific issue with the three joints in the telescopic arm of the robot. When commands are issued for these three joints to move simultaneously, they fail to reach their target positions within a given number of mj_step cycles. Notably, when I issue a command to just one of these joints without involving the others, it reaches its target position accurately and without any issues.
﻿
Additional Details:
﻿
The problem only arises when the telescopic arm's joints are commanded to move simultaneously.
Each joint individually can reach its assigned target when operated alone.
This issue does not affect other types of joints in the robot, such as those used for rotation and translation.
Request for Assistance:
Could you please provide any insights or suggestions on what might be causing this issue and how to resolve it? Are there specific settings or adjustments within MuJoCo that could help ensure all joints in the telescopic arm reach their targets simultaneously?
﻿
I am looking forward to your expert advice to help improve the functionality of this robotic model.
﻿
Thank you for your assistance.

```XML
<?xml version="1.0"?>
<mujoco>
  ......
  <worldbody>
    <body name="base_body">
      <!-- Frame -->
      <body name="frame1">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66"/>
        <geom name="frame1" type="mesh" mesh="frame1" pos="0 0 0" contype="1" conaffinity="1"/>
      </body>
      <body name="frame2">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66"/>
        <geom name="frame2" type="mesh" mesh="frame2" pos="0 0 0" contype="1" conaffinity="1"/>
      </body>
      <body name="frame3">
        <inertial diaginertia="222.35 143.37 209.67" mass="1e-2" pos="0 0 0.66"/>
        <geom name="frame3" type="mesh" mesh="frame3" pos="0 0 0" contype="0" conaffinity="0"/>
        <body name="Y_linear" pos="0 0 1.137">
          <inertial diaginertia="1e-4 1e-4 1e-4" mass="1" pos="0.01 -0.02 -0.1"/>
          <geom name="Y_linear" type="mesh" mesh="Y_linear" pos="0 0 0" contype="0" conaffinity="0"/>
          <joint name="Y_linear_joint" type="slide" axis="0 -1 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.345 0.345"/>
          <body name="X_linear" pos="0 0 -0.002">
            <inertial diaginertia="1e-4 1e-4 1e-4" mass="1" pos="-0.04 -0.01 -0.2"/>
            <geom name="X_linear" type="mesh" mesh="X_linear" pos="0 0 0" contype="1" conaffinity="0"/>
            <joint name="X_linear_joint" type="slide" axis="1 0 0" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="-0.225 0.281"/>
            <body name="Z1_linear" pos="0 0 0">
              <inertial diaginertia="1e-4 1e-4 1e-4" mass="1" pos="-0.04 0 -0.24"/>
              <geom name="Z1_linear" type="mesh" mesh="Z1_linear" pos="0 0 0" contype="1" conaffinity="0"/>
              <joint name="Z1_linear_joint" type="slide" axis="0 0 -1" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="0 0.276"/>
              <body name="Z2_linear" pos="0 0 0">
                <inertial diaginertia="1e-4 1e-4 1e-4" mass="1" pos="-0.01 0 -0.25"/>
                <geom name="Z2_linear" type="mesh" mesh="Z2_linear" pos="0 0 0" contype="1" conaffinity="0"/>
                <joint name="Z2_linear_joint" type="slide" axis="0 0 -1" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="0 0.325"/>
                <body name="Z3_linear" pos="0 0 0.207">
                  <inertial diaginertia="1e-4 1e-4 1e-4" mass="2" pos="0 0 0"/>
                  <geom name="Z3_linear" type="mesh" mesh="Z3_linear" pos="0 0 0" contype="1" conaffinity="0"/>
                  <joint name="Z3_linear_joint" type="slide" axis="0 0 -1" pos="0 0 0" damping="0.1" frictionloss="0.1" limited="true" range="0 0.207"/>
                  ......
  </worldbody>
  <actuator>
    ......
    <position name="Z1_linear_position" joint="Z1_linear_joint" ctrllimited="true" ctrlrange="0 0.276" kp="1600" kv="30"/>
    <position name="Z2_linear_position" joint="Z2_linear_joint" ctrllimited="true" ctrlrange="0 0.325" kp="1500" kv="30"/>
    <position name="Z3_linear_position" joint="Z3_linear_joint" ctrllimited="true" ctrlrange="0 0.207" kp="1100" kv="30"/>
    ......
  </actuator>
</mujoco>
```
python files
`        joint_ids = np.array(
            [mujoco.mj_name2id(self.model, mujoco.mjtObj.mjOBJ_JOINT, name) for name in joint_names])
        for _ in range(50):
            mujoco.mj_step(self.model, self.data)`

erez-tom, 2024-05-11:
Thanks for reaching out.  Sadly, I don't understand your issue. How are you
"issuing commands"? How can you "issue commands" to only one joint at a
time? The entire data.ctrl vector is read at every time step!

yuvaltassa, 2024-05-12:
You should look at the [Hello Robot Stretch 2](https://github.com/google-deepmind/mujoco_menagerie/tree/main/hello_robot_stretch#readme) in the Menagerie. It has a telescopic arm.

closed 2024-05-12 by yuvaltassa