google-deepmind/mujoco #1647: Error: engine error: edge ordering is incoherent between flex and plugin

gpans, 2024-05-10:
Hi,

I am a masters student trying to design a soft locomotive robot using Mujoco for my thesis. I've just started with some .obj files and the elasticity.solid plugin using flexcomp type mesh. However, when I do so i get the following error: 

Error: engine error: edge ordering is incoherent between flex and plugin

My xml code is as follows:

<mujoco> 
  <option gravity = "0 0 -9.81" integrator="implicitfast" solver="Newton" tolerance="1e-6" timestep=".001"/>
    
  <include file="environments/scene.xml"/>

  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>

  <compiler meshdir="asset\" />

  <compiler autolimits="true"/> 

  <size memory="150M"/>

  <visual>
    <map stiffness="100"/>
  </visual>
 

  <worldbody>
    <flexcomp name="softlimb" dim="2" type="mesh" file="soft_limb_1.obj">
      <plugin plugin="mujoco.elasticity.solid">
        <config key="poisson" value="0.3"/>
        <config key="young" value="1e8"/>
        <config key="damping" value="0.002"/>
      </plugin>
    </flexcomp>
  </worldbody>
</mujoco>
  

I have tried loading the file through xml as a rigid mesh body and that was no issue (see below).

[image]

 Any guidance or tips would be much appreciated 
Thanks 
Gaetan

quagla, 2024-05-28:
Could you please attach the OBJ file that it's causing the issue?

gpans, 2024-05-28:
[soft_limb_1.zip](https://github.com/google-deepmind/mujoco/files/15467143/soft_limb_1.zip)

Hi, it wouldn't allow .obj files so I have zipped it. Thanks

shunsei-t, 2024-06-05:
[model.zip](https://github.com/user-attachments/files/15573554/model.zip)
I'm trying to use these models, but it's not working. `mujoco.elasticity.solid`