google-deepmind/mujoco #1648: Unity error: unrecognized attribute: 'sensornoise'

hubbardp, 2024-05-10:
Microsoft Windows 11 Enterprise, 10.0.22631 Build 22631
Downloaded the latest MuJoCo release: https://github.com/google-deepmind/mujoco/releases/tag/3.1.5
Extracted to C:\Users\hubbardp\MuJoCo
Cloned https://github.com/google-deepmind/mujoco.git
Used Unity Hub to create a new project with Unity editor 2022.3.26f1
Used the Unity editor "Window/Package Manager" to open "unity/package.json" from the cloned repo
The package loaded successfully, describes itself as version 3.1.6
Used the Unity editor "Assets/Import MuJoCo Scene" to load "model/humanoid100/humanoid100.xml" from the cloned repo
The scene appears, with the humanoid standing on a platform surrounded by little colored spheres and cubes
Press the Unity editor "Play" button

IOException: Error loading the model: XML Error: Schema violation: unrecognized attribute: 'sensornoise'

Element 'flag', line 1

Mujoco.MjVfs.LoadXML (System.String filename) (at C:/Users/hubbardp/Documents/mujoco-bug/mujoco/unity/Runtime/Tools/MjVfs.cs:83)
Mujoco.MjEngineTool.LoadModelFromString (System.String contents) (at C:/Users/hubbardp/Documents/mujoco-bug/mujoco/unity/Runtime/Tools/MjEngineTool.cs:69)
Mujoco.MjScene.CompileScene (System.Xml.XmlDocument mjcf, System.Collections.Generic.IEnumerable`1[T] components) (at C:/Users/hubbardp/Documents/mujoco-bug/mujoco/unity/Runtime/Components/MjScene.cs:179)
Mujoco.MjScene.CreateScene (System.Boolean skipCompile) (at C:/Users/hubbardp/Documents/mujoco-bug/mujoco/unity/Runtime/Components/MjScene.cs:171)
Mujoco.MjScene.Start () (at C:/Users/hubbardp/Documents/mujoco-bug/mujoco/unity/Runtime/Components/MjScene.cs:89)

Balint-H, 2024-05-13:
Hello, thanks for the report, this is being fixed at the moment and the PR for it should be finished soon.

hubbardp, 2024-06-11:
Seems to be working again now with MuJoCo release 3.1.6 and the version of the Unity plugin whose `package.json` has `"version": "3.1.7"`.

closed 2024-06-11 by hubbardp