google-deepmind/mujoco #1651: Derive xpos from qpos

amaulap2, 2024-05-12:
Hi,

I'm a student and I'm trying to use MuJoCo for a Semester Project.

I'm looking for some help with the xpos derivation.

My project consists in using Neural Network to update the control parameters of a Mujoco model. For that I need to backpropagate through mujoco. So the idea I had is to use a linearization of the mujoco steps to easily backpropagate through it. So I used mj_transition_FD function to get the A and B's. It's working perfectly for now. 

The problem that I get is that, I am continuing an existing project and in this project the loss used for backpropagating are using xpos, that I can only have by simulating with a step or something else but when using the linearization I only have the qpos. I don't want to change that much the existing code it will be too much effort I think.

So my question is, is there an easy way to get the xpos from the qpos ? 

Because I saw on an other issue from May 20, 2023 (https://github.com/google-deepmind/mujoco/issues/889):
That : "xpos is a derived quantity that is computed from qpos during mj_step" 

But I've been trying to look for the function mj_step in the github or an explanation but I haven't succeed.

Hope that my message is clear and that some of you could help me on that.
Thank you very much.

yuvaltassa, 2024-05-12:
The function that computes xpos from qpos is mj_kinematics. You can also get the analytical Jacobian of this function using mj_jac and related functions.

closed 2024-05-12 by yuvaltassa

amaulap2, 2024-05-12:
Frist wow and thank you for the rapidity of reaction, but then if I use mj_kinematics I will lose track of my gradient calculation. So probably I should go for mj_jac but I am not sure to understand how to use that ? Because I haven't used it to go from qpos to xpos up to now.

Thank you again.

WestlifeX, 2024-09-03:
Hi, I'm having the same problem as you, have you already solve this problem?

amaulap2, 2024-09-03:
Hi, i did something that worked but I wouldn't say that it is solved ahah.

The system that I was using was pretty simple geometrically speaking. So i simply did some trigonometry to create a function that passes from xpos to  qpos. But as it is only pojection, it works only for my system as it is now.