google-deepmind/mujoco #1655: [MJX] Model mesh_vert no longer jax array

areiner222, 2024-05-13:
I noticed that the mesh vertices (previously geom_convex_verts) of the mjx.Model are no longer jax arrays. I think this precludes me from having different mesh vertices over my vmapped batch dimension which is what I've done historically - use shape parameters to modify a set of convex meshes and recompute inertial properties, etc in jax.

btaba, 2024-05-13:
Hi @areiner222 , this got refactored in a large change where we added `condim`, but I plan on adding these fields back the the `mjx.Model` at `mjx.put_model` time. Apologies for the inconvenience in the interim

areiner222, 2024-05-13:
Thanks for the update @btaba. 

Yes will pin a prior version for now  - 3.1.3 works just fine and will try 3.1.4. 
I was looking to upgrade in order to be able to try higher condim to improve stability for "mesh feet/hands" x floor plane collisions. Can hold off on that until the the fields are added back.

btaba, 2024-05-14:
Hi @areiner222 , we wouldn't perform the mesh "processing" in jax, as you noted jax puts a very hard limitation on implementing any of those processing functions. We'd precompute and store results in `mjx.Model` as `jax.Array`

commit c511d022 (2024-05-29): Add hfield. Fixes #1655 #1491 #1695

PiperOrigin-RevId: 638447127
Change-Id: Ib1e5020a8407bc100145a6b382e985c03dd4a848
closed 2024-05-29 by copybara-service[bot]