google-deepmind/mujoco #1656: mujoco display irregularity

ylangyalng, 2024-05-13:
Hi,

I'm a student and I'm trying to use MuJoCo for RL training

I'm looking for some help with display issue

I tried mujoco 200, 237, 315, None of them work properly on top of my laptop :(
All of them are only showing a little bit of the image in the top left corner as shown below
It seems to be just a display issue, I can even manipulate the options in this small area

And I use a computer with an amd cpu with core graphics and a nvidia graphics card, I don't know if that helps.
I'm hoping to find a way to get it to work on my computer, thank you very much for your reply!

[image]

ylang-ylang, 2024-05-13:
Never mind.I updated my core graphics card driver and everything seems to be working fine! :)
[image]
For those of you who also using amd cpu with Radeon 680M core graphics, updating to 31.0.24031.5001 will solve your problem!

ylang-ylang, 2024-05-13:
After I restarted the laptop, the strange thing happened again
[image]
I really don't have a clue.

RagibRownak, 2024-06-14:
Same issue there also. Did anyone found any solutions for this issue?

yuvaltassa, 2024-06-15:
Hmm there used to be an open issue about problems related to AMD GPUs but I can't seem to find it...

@saran-t can you find it?

RagibRownak, 2024-06-19:
is there any way to solve this gui interface issue with AMD GPUs? Any advice or hack or solve is appreciated.

wjxway, 2024-07-07:
https://github.com/google-deepmind/mujoco/issues/639

kevinzakka, 2024-08-30:
Let's keep track of AMD related issues in #639.

closed 2024-08-30 by kevinzakka